PE5PVB/TEF6686_ESP32 #10: Support for German Ö and Ä in RT

andimik, 2023-06-09:
In case there is a special character in the Radiotext, this will be skipped (even for the following ones)

[image]
[image]

The interesting thing is, that the TEF shows the character `ü` but not the other ones like `ä`.

`Nächste Sendung: Journal um fünf` gets `Nste Sendung: Journal um fünf`

`Im Gespräch mit der` gets `Im Gespr mit der`

[image]
[image]

PE5PVB, 2023-06-09:
Thanks for the feedback. I'm currently working on the EBU characterset but wan't able to test it. This will be fixed soon.

andimik, 2023-06-09:
[17fe6d6](https://github.com/PE5PVB/TEF6686_ESP32/commit/17fe6d6a89bb38f03073d596268e5aebcc6d87cb) is a good start.

Anyway, nearly all Radiotext characters with Umlaut are cropped, only `ä` is replaced by a placeholder now.

RDSSpy has logged that (BER = 0)

```
00:41:54 dem genialen Jazzgitarristen George Benson                      
00:42:24 Ö1 Service: Tel. (01) 501 70 371 (Mo-Fr, 8-21 Uhr)              
00:42:54 Nächste Sendung: Nachrichten                                    
00:43:24 Fragen zum Ö1 Programm? Tel. (01) 501 70 371                    
00:43:55 Jetzt in Ö1: George Benson: Das Timing unbewusster Grazie       
00:44:24 (Fortsetzung). Die Spielräume Nachtausgabe verbeugt sich vor    
00:44:55 dem genialen Jazzgitarristen George Benson                      
00:45:25 Ö1 im Internet: oe1.ORF.at                                      
00:45:54 Nächste Sendung: Nachrichten                                    
00:46:25 Ö1 Club: (01) 501 70 370 (Mo-Fr, 9-17 Uhr)                      
00:46:54 Jetzt in Ö1: George Benson: Das Timing unbewusster Grazie       
```

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

andimik, 2023-06-10:
Good news: I found the bug in https://github.com/PE5PVB/TEF6686_ESP32/commit/17fe6d6a89bb38f03073d596268e5aebcc6d87cb

You were using wrong hex numbers.

For example capital `Ö` (like for Österreich) is defined in https://github.com/PE5PVB/TEF6686_ESP32/blob/17fe6d6a89bb38f03073d596268e5aebcc6d87cb/src/TEF6686.cpp#L528 

but it should be

```
	case 0xd7: return 0x4f; break;  // &Ouml;   Ö
```

[image]
(Source EN 50067)

andimik, 2023-06-10:
[image]

PE5PVB, 2023-06-10:
Did you change and test it?

andimik, 2023-06-10:
Yes, for this screenshot I've changed it, compiled it and uploaded it to the TEF.

According to the RDS specification the following characters are allowed (starting with 0x30):

```
0123456789:;<=>?
@ABCDEFGHIJKLMNO
PQRSTUVWXYZ[\]―_
‖abcdefghijklmno
pqrstuvwxyz{|}¯ 
áàéèíìóòúùÑÇŞβ¡Ĳ
âäêëîïôöûüñçşǧıĳ
ªα©‰Ǧěňőπ€£$←↑→↓
º¹²³±İńűµ¿÷°¼½¾§
ÁÀÉÈÍÌÓÒÚÙŘČŠŽÐĿ
ÂÄÊËÎÏÔÖÛÜřčšžđŀ
ÃÅÆŒŷÝÕØÞŊŔĆŚŹŦð
ãåæœŵýõøþŋŕćśźŧ 
```
(many thanks to https://github.com/windytan/redsea/blob/b34fd7ce72516d94407957a7fdb3e0e8eef751df/src/rdsstring.cc#L34 )

[image]

see attached spreadsheet: 
[rds_codepages.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/11712058/rds_codepages.zip)

For example, `Ö` is 0xD7 in RDS, but 0xD6 in utf8, see https://www.utf8-zeichentabelle.de/unicode-utf8-table.pl?number=1024

**Could you tell me which code page for Aura2CondensedPro is used on the TEF screen?** Then I can map it and send you a pull request if you want.

The font works (at least) with umlauts, see `ü` I got last time. So I think we don't need to convert it to another character.

```
#define FONT24 &Aura2CondensedPro_Regular24pt7b
#define FONT14 &Aura2CondensedPro_Regular14pt8b
#define FONT7  &Aura2CondensedPro_Light7pt8b
```

Because I tried the following without success ...

`case 0xd7: return 0xC396; break;   // &Ouml;   Ö`
and
`case 0xd7: return 0xd6; break;   // &Ouml;   Ö`
and
`case 0xd7: return 0x96; break;   // &Ouml;   Ö`

[image]
(sorry for the bad quality, it should be `Nächste Sendung: Tipps für Ö1 Club-Mitglieder` in German)

PE5PVB, 2023-06-10:
It's not that easy. RDS comes as UTF8, but the font uses extended ascii. Currently I masked the öïëä as from your table, but it should be nice to have a MPX recording of that Austrian station. Please make a recording via a virtual audio cable from your SDR of about 10 minutes in WAV at 192kHz and post it via wetransfer.

andimik, 2023-06-10:
Tomorrow I will add all other characters and test with Austrian and Slovene stations here. Maybe I find Italian channels with special characters, too.

It's a good opportunity as I am living next to two borders...

PE5PVB, 2023-06-10:
Well, if you can help me with MPX recordings you are very welcome. In that way I can easily debug.

andimik, 2023-06-10:
Sure, I will send you the link to the recordings then. Any specific format? I suppose 192 kHz 16bit mono PCM.

PE5PVB, 2023-06-10:
Yes, correct, uncompressed WAV. Ofcourse the full audio range, so I can put it on my test transmitter ;-)

andimik, 2023-06-11:
In the following MPX file (sampled in 192k, hopefully in mono) you will find the following **radiotext messages**:

```
Drei Kandidatinnen und Kandidaten aus ganz Österreich lösen
Bernhard Fellinger Rätsel aus den Bereichen Kultur, ...
Ö1 Club: (01) 501 70 370 (Mo-Fr, 9-17 Uhr)
Nächste Sendung: Nachrichten
Ö1 Literaturlisten: Tel. (01) 501 70 371
Jetzt in Ö1: Das Ö1 Quiz mit Bernhard Fellinger
Drei Kandidatinnen und Kandidaten aus ganz Österreich lösen
Bernhard Fellinger Rätsel aus den Bereichen Kultur, ...
E-Mail an Ö1: oe1.service@orf.at
Nächste Sendung: Nachrichten
```

You can read this with 

```
cat /tmp/a201_oe1_192k_mono.rec | redsea -E -r 192k -p | grep text | jq -r ".partial_radiotext" 
```

The file has been created with 

```
rtl_fm -M fm -l 0 -A std -p 0 -s 192k -g 30 -F 9 -f 92.8M | tee /tmp/a201_oe1.rec | redsea -E -r 192k -p | grep text
```
# link to the file

[https://we.tl/t-ZNVKooPIKt](https://we.tl/t-ZNVKooPIKt?utm_campaign=TRN_TDL_05&utm_source=sendgrid&utm_medium=email&trk=TRN_TDL_05)

PE5PVB, 2023-06-11:
I can't play this file. I need .WAV format so I can playback it on my test transmitter.

andimik, 2023-06-11:
Then you should convert it with `sox` into the right format or tell me the `rtl_fm` parameters.

PE5PVB, 2023-06-11:
Usually I use a virtual audio cable and just record it with Audicity. I have no knowledge of Linux, sorry.

PE5PVB, 2023-06-11:
If you can convert it, please do so.

andimik, 2023-06-11:
Oh, did not know it.

The output should be `signed 16 bit ints`, so it's basically `wav`. Just try to rename it.

PE5PVB, 2023-06-11:
Tried that, didn't work.

andimik, 2023-06-11:
ok, let me boot into Windows :-)

PE5PVB, 2023-06-11:
Conversion worked. I can now play the file. Thanks for your help

andimik, 2023-06-11:
In https://github.com/walczakp/rds-spy-logs I found:

```
Bäst musik just nu!
Das Ö1 Tagesprogramm: (01) 501 70 371
FONK! Det er lørdag
Förfest med Richard Herrey
HITRADIO VNıOCINA - RADIO KTERE HRAJE
Hörerservice: 0800-218 16 16
ICOTOVI ZOBI,  BIG BAND RTV SLOVENIJA - LEŽERNO JE MODERNO
Jetzt in Ö1: Live von den Salzburger Festspielen - Wolfgang ...
Klingan: Mixtejp med gäst-DJ
Kostenloses Hörertelefon: 0800 / 5900 111
Nächste Sendung: Tipps für Ö1 Club-Mitglieder
Næste: P3 Nyhede
Næste: P3 Nyheder
Næste: Radioavisen
Ö1 Service: Tel. (01) 501 70 371 (Mo-Fr, 8-21 Uhr)
P3 Musikdokumentär
Več kot radio

```

with 

```
/tmp/rds-spy-logs$ cat  */*.spy | redsea -E -h | grep text | jq -r ".radiotext" | sort | uniq | grep --color='auto' -P '[^\x00-\x7F]' 
```

PE5PVB, 2023-06-11:
We made a fontset based on the RDS charset you provided. But it seems not to work on all characters. Needs some more investigation, or a direct converter to UTF16. Will be investigated.

andimik, 2023-06-11:
I got it working (at least for German characters) on the TEF, but not live with Austrian radio stations.

My script using https://github.com/ChristopheJacquet/PiFmRds/ added another byte (by mistake), and the second byte was correct (like `ö`) on the display.

I guess the actual problem is with int16 and int8.

andimik, 2023-06-12:
Is that helpful?

https://playground.arduino.cc/Main/Utf8ascii/

PE5PVB, 2023-06-14:
Can you check RDS again?

closed 2023-06-14 by andimik