PE5PVB/TEF6686_ESP32 #9: RT and PS show wrong characters

andimik, 2023-06-09:
From June commits on, the PS and RT on the TEF sometimes show wrong characters (even displayed in yellow letters). Till 30th May (1.16RC) it was correct.

See https://github.com/PE5PVB/TEF6686_ESP32/commit/825cc683397269375122b15d9cf8a7f73078f58b

For example the correct PS

> RADIO-K

gets

> RADIO,?

Better to show nothing rather than garbage ...

see screenshots.

[image]
[image]
[image]

Testing with latest commit:

```
$ git show
commit f926634f0152cd8c7e7e718ada4cad4aff3522f5 (HEAD -> main, origin/main, origin/HEAD)
Merge: dac00bc e1add2c
Author: Sjef Verhoeven PE5PVB <pe5pvb@het-bar.net>
Date:   Fri Jun 9 12:51:14 2023 +0200
```

PE5PVB, 2023-06-09:
Will be fixed soon.

PE5PVB, 2023-06-11:
Closed, see other issue.

closed 2023-06-11 by PE5PVB

andimik, 2023-06-11:
But this is not fixed yet, you just have closed by mistake, right?

I'm still getting wrong characters.

This is **NOT** related to special characters. I'm talking about ASCII characters and PI-Codes which vary due to bad reception.

PE5PVB, 2023-06-11:
I know, but there is already a topic on RDS texts