PE5PVB/TEF6686_ESP32 #33: Sudden reboot during RDSSPY WiFi connection

andimik, 2023-06-18:
During an RDSSPY connection via WiFi the TEF rebooted suddenly.

After this reboot RDSSPY received no data at all. I have switched it off an on several times to get it working again.

This sudden reboot happened a second time some minutes later.

I have no idea what went wrong here.

Using e2cc859, old Win 7 laptop 32 bit. WiFi is my hotspot on the mobile next to the TEF.

PE5PVB, 2023-06-18:
Is it possible your WiFi signal is a bit low?

andimik, 2023-06-18:
You mean a low signal caused the reboot?

PE5PVB, 2023-06-18:
It can be possible. I still have to build in a check, when wifi is lost, servers should stop.

andimik, 2023-06-18:
Yes, but I expected resuming after having WiFi again. This seems not to be the case.

PE5PVB, 2023-06-18:
Not possible, the socket is closed then.

andimik, 2023-06-18:
Understood. But what do you recommend next?

I have tried

- switching WiFi off and on
- restarting tuner
- setting mode to xdr-gtk and back to rdsspy

I can't tell you what has helped (, luckily) at the end, but the TEF should at least tell the user that the connection has been lost and propose the next step.

And we shall not have a reboot here.

PE5PVB, 2023-06-18:
Yes I know. But has not been implemented yet. This is still a development version.
Today I rewrote the whole RDS part.

andimik, 2023-06-18:
Ok, will test it in the next days.

PE5PVB, 2023-07-21:
I modified the software to switch off the brownout reset from the ESP32. This was sometimes triggered due design flaws in the Chinese receivers.

closed 2023-07-21 by PE5PVB