PE5PVB/TEF6686_ESP32 #65: Translation for NOT POSSIBLE

andimik, 2023-07-05:
There is still a text which could be localized:

`NOT POSSIBLE` 
in line 1223 and 1226

NoobishSVK, 2023-07-06:
I would probably change this to "DISCONNECT USB", just like in my fork of the FW as that would tell the users what to do instead.

PE5PVB, 2023-07-06:
Fixed. Please test and let me know.

andimik, 2023-07-06:
yes, now it's localized, but once the text has been shown, some pixels below the (former) text become black. For instance half of the letter `M` in `MHz` is now hidden till a restart.

[image]

PE5PVB, 2023-07-06:
Ah ok. This has to do with alligning. I'll look into this.

PE5PVB, 2023-07-10:
I checked this, but can't reproduce the problem.

andimik, 2023-07-10:
Ok, will try with the latest build tomorrow

closed 2023-07-21 by PE5PVB