PE5PVB/TEF6686_ESP32 #128: Empty AF does not show text 87

andimik, 2023-07-26:
Other than EON or RT+, empty AF will not tell, that there is no AF. 

For example, this station has no AF:
http://andimik.bplaced.net/sendinfo.php?id=256

[image]

Expectation: 
Write the text defined in 87.

https://github.com/PE5PVB/TEF6686_ESP32/blob/921b04170a72e9fe7cae6180f5d3833f370c699d/src/language.h#L91

Could you check
https://github.com/PE5PVB/TEF6686_ESP32/blob/921b04170a72e9fe7cae6180f5d3833f370c699d/TEF6686_ESP32.ino#L3239

HyperDX, 2023-07-26:
I noticed that if the frequency that you are tuned into is broadcasted on AF list by radio station it somehow doesn't appear on AF list on the screen. Maybe this issue is related to it.

andimik, 2023-07-26:
I don't think so. As there is no AF transmitted, the TEF cannot check that.

HyperDX, 2023-07-26:
I've done a quick test. I broadcasted RDS on 87.7 MHz. At a first attempt empty AF list in RDS encoder settings.
TEF shows "No AF available". Second try I put 87.7 MHz on AF list and AF list is blank as shown on the first screen.

PE5PVB, 2023-07-27:
The transmitter frequencies will not be added to the AF list. I filtered that one.

PE5PVB, 2023-07-27:
Ok. What is the actual problem? When the only AF is the current frequency, it will be filtered. When I check AF with transmitters which transmit multiple frequencies, I see no problems.

PE5PVB, 2023-07-27:
The reason why it shows empty is because AF flag is there, but the only frequency in the AF list is it's own frequency, which is not shown.

andimik, 2023-07-27:
Let me check in the next days. On my link you can see that AF is not used. Maybe the broadcaster has changed it in the meantime. 

I am aware that TEF won't show the received frequency. 

In my opinion it was **not** the case as I assumed that there are no AF transmitted at all at this station.

HyperDX, 2023-07-28:
> Ok. What is the actual problem? When the only AF is the current frequency, it will be filtered. When I check AF with transmitters which transmit multiple frequencies, I see no problems.

I'd say it would be important for person who manages online RDS databases where they need to fill AF list. I think on the other hand for regular person it doesn't matter is it filtered or not.

andimik, 2023-07-29:
Closed. The mentioned radio station has added its only frequency to AF in the meantime.

closed 2023-07-29 by andimik