PE5PVB/TEF6686_ESP32 #129: PTYN only shows 4 characters

andimik, 2023-07-26:
Described in https://discord.com/channels/1053804249651359765/1080501413651693578/1133824988483440640

PTYN is specified for 8 characters, but it will only show the first 4.

HyperDX, 2023-07-26:
Also there should be tag "PTYN:" before it. Because now we don't know what this field is if PTYN is displayed. If no PTYN is available we have "PTYN N/A" info which is corrent.

PE5PVB, 2023-07-27:
Can you send me a MPX recording as WAV 192kHz? I can't reproduce this problem.

@HyperDX PTYN: with a 8 chars text doesn't fit in the window.

PE5PVB, 2023-07-27:
Fixed.

closed 2023-07-27 by PE5PVB

andimik, 2023-07-27:
Regarding an MPX

The nearest (Italian) station with PTYN is one hour drive from here...

Maybe another can record that.

PE5PVB, 2023-07-27:
No need. I already fixed it.