PE5PVB/TEF6686_ESP32 #131: Squelch shows double text

andimik, 2023-07-26:
The word `OFF` (in the pic it's the localized German word `AUS`) is in the background of the squelch value.

[image]

This seems to happen when the squelch is changed during scanning.

PE5PVB, 2023-07-27:
I was not able to reproduce it. Can you make a video where you reproduce this problem?

MCelliotG, 2023-07-27:
In my opinion, the value should move to the same line as SQ: It causes issues in translations where the word for OFF is more than 3 characters.

PE5PVB, 2023-07-27:
I have changed the alligment of the text, this will help. There is not enough space on the screen to do it in one line.

PE5PVB, 2023-07-27:
Fixed

closed 2023-07-27 by PE5PVB