PE5PVB/TEF6686_ESP32 #132: MEM places (are not band-related)

andimik, 2023-07-26:
Although the description changed from `FM` to `OIRT`, the MEM places are not related to each band. I know that this is a limitation, but the MEM position after the BAND is misunderstanding.

So a user will overwrite the MEM position 1 although he thinks, this is free.

Suggestions: the text for BAND and MEM place should not be displayed one after the other. See advanced RDS view.

[image]

ohmytime, 2023-07-27:
In the next function ,I may add MEM Pages, I can change it's location.

PE5PVB, 2023-07-27:
I changed the indicators from position, but what is the real question? Memory channels are currently designed to have a mix of all bands.

mx4zxc, 2023-07-27:
I was wondering the same thing. But what is the plan in the future? Mix with all bands or separate them between each band? I'm super happy with storing frequencies in the memory, but I think it might be a little easier if the memory channels were split up into each band?

andimik, 2023-07-27:
> but what is the real question? Memory channels are currently designed to have a mix of all bands.

It's a kind of usability. OIRT1 is the same as FM1...

PE5PVB, 2023-07-27:
Yes, the plan was a global channellist. Please start a discussion about it in Discord, I never use memory channels, so ask others opinions.

closed 2023-07-27 by PE5PVB