PE5PVB/TEF6686_ESP32 #133: Wrong calculation for SNR on OIRT

andimik, 2023-07-26:
My antenna cable was not properly connected, so I got these values

[image]

Switching to FM the values were ok, so only OIRT affected.

BTW: Have not tried to get that behavior a second time.

HyperDX, 2023-07-26:
Also it can be related to AM where SNR measurements make no sense

ohmytime, 2023-07-27:
> Switching to FM the values were ok, so only OIRT affected.
Is the current frequency a real valid station? I tested with latest code , it works fine on OIRT band on any frequency.
Or you can press Rotary button + BW button to restore to factory .
Then have a try.

ohmytime, 2023-07-27:
> Also this issue can be related to AM where SNR measurements make no sense

SNR not correct is an known issue. to be done.

PE5PVB, 2023-07-27:
Signalstrength if -99.8dBuV is not possible. I've seen this behaviour before some time ago. It was caused by wrong data from the TEF chip. I've added a constrain between -20 and +120 dBuV for both AM and FM