PE5PVB/TEF6686_ESP32 #134: Flipped rotary for BAND selection on SW

andimik, 2023-07-26:
The rotary on SW for the BAND selection is flipped. I guess it is checking the meters, instead of the frequency.

For example if you want to switch from 49m to 41m band you need to scroll down instead of up.

ohmytime, 2023-07-27:
Thank you for your feedback, The previous direction was based on MI BAND, but later I adjusted it to frequency
If in MAN mode, the frequency increases, Then in BAND mode, the frequency also increases

closed 2023-07-27 by ohmytime

Mike86-CZE, 2023-08-10:
Could it at least be made optional?  I don't like it too much either.

andimik, 2023-08-10:
@Mike86-CZE

As I don't use AM at all, could you test if the direction is OK now?

Mike86-CZE, 2023-08-10:
By turning forward, the frequency decreases and the meters increase. Originally it was the other way around, that seems better to me.