PE5PVB/TEF6686_ESP32 #135: Offset on OIRT frequencies

andimik, 2023-07-26:
@ohmytime 

Dear Kevin, the OIRT frequencies are wrong. There is an offset as the edges are not correct.

I've changed them to

```
#define FREQ_FM_OIRT_START         6590   // use values of 1/10 * kHz
#define FREQ_FM_OIRT_END           7400   // use values of 1/10 * kHz
```
And it works!

[image]

Due to the new OIRT band implementation, there is no need to have a menu item for the FM lower edge in my opinion.

ohmytime, 2023-07-27:
> I've changed them to
> 
> ```
> #define FREQ_FM_OIRT_START         6590   // use values of 1/10 * kHz
> #define FREQ_FM_OIRT_END           7400   // use values of 1/10 * kHz
> ```
> 
> And it works!
> 

Thank you for your feedback, you can submit a PR to change them directly.

> Due to the new OIRT band implementation, there is no need to have a menu item for the FM lower edge in my opinion.

Well, I think OIRT and FM can be different bands, because some users would hide OIRT and keep FM band.
So I think it is better to keep this function, so that they can customize FM band range.

PE5PVB, 2023-07-27:
Changed in repository.

closed 2023-07-27 by PE5PVB

MCelliotG, 2023-07-27:
FM lower edge should stay, just to cover 74 and up, in order to still include Japan FM.