PE5PVB/TEF6686_ESP32 #142: The last characters of the AF list on the RDS screen are missing.

Mike86-CZE, 2023-07-27:
On the advanced RDS screen, when the AF list is scrolled, the last few characters are not displayed.  For example, the last frequency in the list is "106.1", but only "10" is displayed. It happens often, but not always.

andimik, 2023-07-27:
Can you post a screenshot?

Mike86-CZE, 2023-07-27:
https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/c5989fca-3033-4218-864b-f1a25936964c

PE5PVB, 2023-07-27:
It seems that the AF list is still expanding with new frequencies. This has effect on the ticker, because the end marker is based on the string length. Can you wait for a few minutes after tuning and check if the ticker then works correct?

Mike86-CZE, 2023-07-27:
It keeps happening even after a few minutes.  The signal is good.

Mike86-CZE, 2023-07-27:
Maybe it's happening because of the font.  It would be good to insert spaces after the AF list.

PE5PVB, 2023-07-27:
I've added 5 extra spaces at the end of both AF and EON ticker.

closed 2023-07-27 by PE5PVB