PE5PVB/TEF6686_ESP32 #143: when trying to enter the menu, it displays "not possible"

Mike86-CZE, 2023-07-27:
Sometimes it won't allow me to enter the menu using the middle button and displays "not possible".  This has now happened to me twice in a short period of time.  I still don't know how to target this error.

PE5PVB, 2023-07-27:
Did you connect XDRGTK?

Mike86-CZE, 2023-07-27:
No.  USB or wifi not connected.

PE5PVB, 2023-07-27:
That is strange, because only way this could happen is when XDRGTK was connected (before).

Mike86-CZE, 2023-07-27:
I understand.  But I didn't try to connect at all during the session.  wifi is turned off.

PE5PVB, 2023-07-27:
Maybe something was triggered over USB

ohmytime, 2023-07-27:
Hello, you must hidden some bands. 
Try to enable them.

andimik, 2023-07-31:
Maybe a reset helps here?

Mike86-CZE, 2023-08-04:
The error seems to randomly occur after AF retuning.  I can't say more at the moment. Yesterday's release.

Mike86-CZE, 2023-08-15:
Now this happened to me when saving a preset.  But I can't repeat it.

andimik, 2023-10-10:
https://github.com/PE5PVB/TEF6686_ESP32/blob/cb31950de0c694efdd14714acb28a7a742f128f0/TEF6686_ESP32.ino#L1754-L1760

This is the only code where "Not possible" is displayed. It has to do with xdr-gtk via WiFi or USB.

PE5PVB, 2023-10-10:
Correct, and this should only be called when XDRGTK connection is active. But the issuer said he had no WiFi or USB connected.

The only place where it's set to true in is comms.cpp, and only in specific situations. So I really have no idea.

PE5PVB, 2023-10-10:
Possibly it was fixed during development ;-)
Let's close this one. Please re-open when it occurs again and describe what has been done during power on and the moment the problem occured.

closed 2023-10-10 by PE5PVB