PE5PVB/TEF6686_ESP32 #147: Tickers in advanced RDS screen

lxsxl, 2023-07-27:
The tickers in advanced RDS stop and restart too early and therefore the last characters are (almost) not to be seen. In detail:
EON: The last digit of the last frequency is only to be seen for a few milliseconds, so it should scroll a bit longer even if there is no more information.
AF: Same as for EON
RT: In case of RT with 64 characters the last few characters are not to be seen.
RT+: Actually I couldn't receive RT+ with full 64 characters, maybe it needs also a bit of prolongation.

lxsxl, 2023-07-29:
Todays beta looks fine for the tickers, at least EON and AF. RT I will try to get a full 64 char transmission.

lxsxl, 2023-07-29:
RT is still not visible completely, in my test the last 3 characters were not to be seen.

PE5PVB, 2023-07-31:
I made some improvements on this.

Mike86-CZE, 2023-08-04:
> RT is still not visible completely, in my test the last 3 characters were not to be seen.

I have a similar problem with the latest release.  On both the normal and RDS screens.

andimik, 2023-08-04:
Well, I have some stations which do NOT repeat RT, therefore it might happen that you cannot see the **first** characters, because it already starts scrolling.

Why do we need to have **live** RT tickers? Cached RT messages are good enough.

My proposal: show them after some seconds only (for those without A/B flag) or after A/B flag change.

Mike86-CZE, 2023-08-04:
It would be ideal (at least on the main screen) if the text didn't scroll at all, but fit there entirely.  For example, using a smaller font or a two-line display.