PE5PVB/TEF6686_ESP32 #148: Trigger for TMC flag

lxsxl, 2023-07-27:
I am still curious about TMC response time. This should work much faster. TMC receivers I have seen display the first message after a few seconds.
So I worked on RDS specification (IEC 62106): The presence of a TMC service is signalized in Group 3A (about every five seconds). If there is AID = 0xCD46 in Block 4 this is first strong evidence that TMC is transmitted. To make it more secure check also if group 8A is indicated in block 2 (10000 in the last five bits of block 2). And for perfection check if group 8A is truely transmitted. With this the TMC indicator should respond much faster than actually. Hope that I didn't missunderstood anything...

PE5PVB, 2023-07-27:
Yes I know, but that is more complicated to detect. What I do is check for a specific flag which is only high when TMC is available. But this flag is not transmitted every RDS cycle.

lxsxl, 2023-07-29:
Okay, I understand, to save memory it is enough like it is. But can you tell me what flag this is exactly?

lxsxl, 2023-07-30:
***@***.*** sent you WDR 2 192

1 item, 879 MB in total ・ Expires on 6 August, 2023

Download link:
https://wetransfer.com/downloads/30e56f6a9421972506c9b0348066c72620230730200523/63a15de06702430c222ac2b048b1bd0d20230730200541/51bbdb?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid

1 item

WDR 2 192.wav - 879 MB

[Get more out of WeTransfer, get Pro](https://wetransfer.com/pro?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid)

About WeTransfer: https://wetransfer.com/about
Help: https://wetransfer.zendesk.com/hc/en-us
Legal: https://wetransfer.com/legal/terms

To make sure our emails arrive, please add ***@***.*** to [your contacts](https://wetransfer.zendesk.com/hc/en-us/articles/204909429).

PE5PVB, 2023-07-31:
if (rds.correct && rdsblock == 16 && (rds.rdsB & (1 << 4))) rds.hasTMC = true;                  // TMC flag

closed 2023-07-31 by PE5PVB