PE5PVB/TEF6686_ESP32 #149: Strange behaviour on OIRT band with step size and presets

mrwish7, 2023-07-27:
First of all, I think OIRT being its own band is a very handy solution! It's useful to be able to wrap around on (CCIR) FM from 87.5 --> 108.0 and vice versa.

There seems to be some strange behaviour with the OIRT band in terms of step size and setting presets.

For the tuning steps - if you tune to the top of the OIRT band (74.00 MHz), and wrap around back to the bottom, the steps are incorrect (64.02 ... 66.00 MHz ... etc.) - these are 10kHz too high. If you wrap back in the opposite direction from 64.02 down to 74.00 and tune down, they're back to the correct steps ( 74.00 ... 66.02 ... 65.90). Presumably this is because 74.00 is a correct step so it syncs up in that direction - hopefully the fix for this is as simple as setting the first channel to 64.01 rather than 64.02.

Secondly, if you save a preset on OIRT and then return to it in MEM mode, the signal level suddenly rises to around 115dBµV on the display, and the audio seems to be from an AM signal rather than the OIRT frequency displayed. Cycling back to OIRT via the band selection button clears the error.

I hope this is helpful, let me know if any more details are required! This behaviour has been confirmed by a few different beta testers trying the 27/07/2023 build.

Thanks,
Adam

PE5PVB, 2023-07-27:
I changed lowest frequency to 64.01MHz. About the memory channels, I have to look into this tomorrow.

PE5PVB, 2023-07-28:
OIRT memory channels fixed.

closed 2023-07-28 by PE5PVB