PE5PVB/TEF6686_ESP32 #153: If RDS is lost, there will be no retuning to AF.

Mike86-CZE, 2023-07-28:
When the signal drops rapidly and RDS is lost, retuning to AF will not occur, even though the entire AF list is loaded.

PE5PVB, 2023-07-29:
Fixed.

Mike86-CZE, 2023-07-29:
Unfortunately, the issue still persists in today's release.

Mike86-CZE, 2023-08-06:
Still not work in today's release. Any idea how to fix this?

PE5PVB, 2023-08-07:
Can you check again?

Mike86-CZE, 2023-08-08:
Good job!  Now it already works.

Mike86-CZE, 2023-08-08:
Just a little note.  When RDS is lost, the sound changes its character, as if the tuner is constantly scanning the entire band (known from XDR-GTK), instead of only trying frequencies from the AF list.  I don't know if such behavior is okay.

Mike86-CZE, 2023-08-08:
https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/cbbe8d7b-2038-4cca-8f03-46f537cd6625

PE5PVB, 2023-08-08:
That's correct. It starts scanning all AF frequencies. I'll add 1 second delay between each search attempt.

Mike86-CZE, 2023-08-09:
Ok, that's a good idea.
Would it be possible to have active AF retuning even when the screen is off?

PE5PVB, 2023-08-09:
The screen on/off has no influence on AF tuning.

PE5PVB, 2023-08-09:
Should not be the case. Please test.

Mike86-CZE, 2023-08-09:
The fact is that if I turn on the screen after that, all the RDS have to be reloaded.  Maybe I have something set wrong.

Mike86-CZE, 2023-08-11:
I am now testing it in yesterday's release.  Now when RDS is lost there is no retuning at all.

PE5PVB, 2023-08-11:
Yup. Found and fixed.

Mike86-CZE, 2023-08-14:
> The screen on/off has no influence on AF tuning.

Retuning with the screen off doesn't work because it's probably not just the screen that's turning off, but some other components as well.  This also manifests itself, for example, with XDR-GTK connected via WiFi.  If the screen is turned off based on the screen saver setting, the TEF-PC communication will be interrupted.

PE5PVB, 2023-08-14:
Ah, now I understand what goes wrong. Should be fixed. Please check and report.

Mike86-CZE, 2023-08-15:
Unfortunately.  Retuning still doesn't work after turning off the screen.

PE5PVB, 2023-08-15:
Ah, my mistake.

Mike86-CZE, 2023-08-19:
Is AF selection dependent on signal strength? I have a station tuned in, the AF list is loaded. The signal intensity is 43 dBμV. I start the transmitter on the same frequency and the intensity increases to 58 dBμV. I can hear the tuner trying to search for AF, but it won't switch to one. However, such a situation probably does not occur often in normal practice.

PE5PVB, 2023-08-19:
Yes but noise and multi-path has more weight in scoring the AF

Mike86-CZE, 2023-08-19:
I observed that retuning occurs only when the intensity of the interfering signal drops below approximately 50 dBμV.

Mike86-CZE, 2023-08-20:
It can be said that it is already working well.  After the screen turns off, it still searches for AF even when RDS is lost.  However, if I turn on the screen and there is currently no RDS reception, the AF list is lost and the search does not occur until the AFs are reloaded from RDS.

PE5PVB, 2023-08-31:
I don't understand what you mean. Why RDS reception gets lost after turning on the screen?

Mike86-CZE, 2023-08-31:
RDS reception was lost earlier due to signal drop.  But TEF still knew AF and was trying to find them.  However, when the screen is turned back on, the TEF forgets the AF list and therefore no longer searches for AFs until they are reloaded from the RDS.

PE5PVB, 2023-09-10:
Should be fixed now.

Mike86-CZE, 2023-09-28:
Ok, fixed.  I would just maybe increase the AF search interval to about 5 seconds after several unsuccessful attempts to retune.  Constantly searching for AF after 1 second for a long time feels uncomfortable.

PE5PVB, 2023-09-28:
That should already be the case.

Mike86-CZE, 2023-10-08:
Unfortunately, I have to reopen this issue.  Strange AF retuning behavior after screen off.  In some situations, after turning off the screen, retuning stops working completely.  Other times, after turning off the screen, on the contrary, it tries to retune (but does not retune), even if there is no reason to do so (the signal is good).  After turning on the screen, everything is fine again.

PE5PVB, 2023-10-08:
Ah, found it. Please test again

Mike86-CZE, 2023-10-09:
Now it's fine.  As for the extended AF search interval after several failed attempts, it doesn't seem to be implemented yet.  Even after a long time, he still tries every second.

PE5PVB, 2023-10-09:
What do you mean by that?

Currently AF is triggered when:
1. On RDS loss (1 sec. delay)
2. When signal is very disturbed (1 sec. delay)
3. When signal has little noise (10 sec. delay)

Mike86-CZE, 2023-10-09:
Ah, thanks for the detailed description of AF behavior.  In that case, it would probably be enough to increase the AF search interval when RDS is lost, perhaps to 5 seconds?

PE5PVB, 2023-10-09:
Correct, and that is the way how it should work. It's designed for mobile use. When you are moving closer to another transmitter it will switch. If you should disable it after a few unsuccesfull attempts you will finaly loose reception.

Mike86-CZE, 2023-10-09:
My point is that not every station has such a dense network of transmitters that flawless reception with RDS from at least one transmitter is always available.  If I'm in an area with no RDS reception for an extended period of time, but the sound quality is still pretty good, frequent AF searches are annoying because every attempt to retune is audible.  Currently, this happens every second when losing RDS (but still good sound reproduction), which is annoying.

PE5PVB, 2023-10-09:
Well, it's just a disadvance of a single tuner. I close this issue.

closed 2023-10-09 by PE5PVB