PE5PVB/TEF6686_ESP32 #155: "hijacking" the station

Mike86-CZE, 2023-07-28:
I discovered an interesting bug.  If I have the AF function on and I have a station "ABC" tuned in that temporarily has no other AF available and another station "XYZ" starts playing on most recently tuned frequency over the hill and his RDS data is loaded, the receiver will start retuning to the station's alternate frequencies  "XYZ" and station "ABC" will not return.  This causes the rotary control function to freeze and TEF needs to be restarted.  Unfortunately, I cannot repeat this error at home, but maybe someone with their own transmitter and RDS encoder could do it.

andimik, 2023-07-28:
But I assume that the PI code is different.

Mike86-CZE, 2023-07-28:
Yes, a completely different station, just on the same frequency.

andimik, 2023-07-28:
> station's alternate frequencies "XYZ" and station "ABC" will not return.

I guess this is expected.

> This causes the rotary control function to freeze and TEF needs to be restarted.

Interesting. Maybe I can try with an FM Transmitter.

Mike86-CZE, 2023-07-28:
I managed to repeat the hijacking using a mini FM transmitter.  If an FM radio is connected to it as a sound source, the transmitter also partially transmits RDS data.  This time the controller did not freeze.  However, such behavior of TEF is not good.  The TEF should remember the station originally tuned by the user and ignore incoming RDS data from different PIs.

https://github.com/PE5PVB/TEF6686_ESP32/assets/140615384/1320ca9b-8ed1-489f-84e5-559cbaf619cd

Mike86-CZE, 2023-07-28:
> > station's alternate frequencies "XYZ" and station "ABC" will not return.
> 
> I guess this is expected.
> 

I would rather expect the same behavior as with the car radio.  I tuned in the station "ABC" and I want to listen to it, regardless of the fact that the signals of other stations come in on its frequency while driving.

Mike86-CZE, 2023-07-28:
Maybe we don't understand because of my english from automatic translator. Model situation from the video. I'm on the bus and I want to listen to BLANIK. The bus crosses a hill and behind the hill another station starts playing on the BLANIK frequency, in this case HITRADIO CONTACT. Since BLANIK temporarily has no other available alternative frequency to retune to, TEF remains on the same frequency where another station is already playing. This station (HITRADIO CONTACT) will load its PI code and its AF list and TEF from now search for HITRADIO CONTACT alternative frequencies.  In doing so, I would expect it to still search for the BLANIK alternate frequency I originally tuned in and retune to an available BLANIK alternate frequency as soon as possible.  Hope that's understandable.

PE5PVB, 2023-07-29:
Fixed

closed 2023-07-29 by PE5PVB