PE5PVB/TEF6686_ESP32 #157: Stationlist for Android just one direction

andimik, 2023-07-29:
Seems TEF does not tell Stationlist for Android a changed frequency.

The other way round works, Stationlist for Android controls the TEF.

PE5PVB, 2023-07-29:
Does it work when you use XDRGTK?

andimik, 2023-07-29:
Yes, it does. This was my backup solution as I am currently on a mountain for DXing. But this is PC related of course.

andimik, 2023-07-29:
Can you check if `serial.print` within `if` lines are ever used now?

PE5PVB, 2023-07-29:
Communication to Stationlist is udp traffic, not serial. Do you know a website were communication with the android app is explained?

andimik, 2023-07-29:
Oh, let me then check UDP payload.

andimik, 2023-07-31:
https://dx.3sdesign.de/index.htm?simple_radio_control_protocol.htm

Thanks to a hint of the author I got this link regarding the UDP commands.

PE5PVB, 2023-07-31:
Yes, I used these commands to communicate with Stationlist. Maybe your phone's firewall is blocking something?

andimik, 2023-08-04:
> Does it work when you use XDRGTK?

I fear I did not understand you correctly last week, seems you asked if *Stationlist* via *xdr-gtk* works. I need to try that (see below, with success :-) ).

My answer last week was regarding if *xdr-gtk* works in general.

Today I have tested the first ever 2.00 from 16th June and *Stationlist for Android* (direct connected to TEF via WiFi) is still one direction. Seems this has never worked so far.

But thanks to your question (which I initially understood wrong), **I confirm that *Stationlist for Android* works bi-directional, **when** it's connected to *xdr-gtk*.**

So it's **only broken using direct connection** (when you are **not** able to use your mobile **and** your laptop).

PE5PVB, 2023-08-04:
I changed one thing, the indentification had a space in it, I removed this one, maybe the Android app can't handle this. Please test again with a next beta release.

andimik, 2023-08-04:
Let me guess before I wait for your commit. 

You have replaced the space in `TEF Tuner`.

andimik, 2023-08-04:
I've checked the UDP payload 

# direct from TEF

[stationlist_tef.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12264568/stationlist_tef.zip)

# via xdr-gtk

[stationlist_xdrgtk.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12264600/stationlist_xdrgtk.zip)

Is there a difference?

Maybe yes, in the direct connection there are some malformed packets from AF list.

[image]

PE5PVB, 2023-08-04:
When I open the log from xdrgtk, I still see the log from the TEF tuner?

andimik, 2023-08-04:
[stationlist_xdrgtk_v2.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12264716/stationlist_xdrgtk_v2.zip)

This is now hopefully the right udp packet *Wireshark* capture. You see

```
from=XDR-GTK;RcvLevel=61from=XDR-GTK;RcvLevel=62from=XDR-GTK;RcvLevel=62from=XDR-GTK;RcvLevel=62from=XDR-GTK;RcvLevel=62from=XDR-GTK;RcvLevel=62from=XDR-GTK;RcvLevel=62from=XDR-GTK;RcvLevel=62from=XDR-GTK;RcvLevel=62from=XDR-GTK;RcvLevel=62from=XDR-GTK;RcvLevel=61from=XDR-GTK;RcvLevel=61from=XDR-GTK;RcvLevel=61;bandwidth=-1;freq=99000000from=XDR-GTK;RcvLevel=23;bandwidth=-1;freq=98600000from=XDR-GTK;RcvLevel=20;bandwidth=-1;freq=98400000from=XDR-GTK;RcvLevel=22from=XDR-GTK;RcvLevel=24;bandwidth=-1;freq=98300000from=XDR-GTK;RcvLevel=22
``` 

and the frequency changes from SL for Android

```
from=StationList;freq=99000000from=StationList;freq=98900000from=StationList;freq=98800000from=StationList;freq=98700000from=StationList;freq=98600000from=StationList;freq=98500000from=StationList;freq=98400000from=StationList;freq=98300000from=StationList;freq=98100000from=StationList;freq=98000000from=StationList;freq=97900000from=StationList;freq=97800000from=StationList;freq=97700000from=StationList;freq=97600000from=StationList;freq=97500000from=StationList;freq=97400000from=StationList;freq=97300000from=StationList;freq=97200000from=StationList;freq=97100000from=StationList;freq=97000000from=StationList;freq=96900000from=StationList;freq=96800000from=StationList;freq=96700000from=StationList;freq=96600000from=StationList;freq=96500000from=StationList;freq=96400000from=StationList;freq=96300000from=StationList;freq=96200000from=StationList;freq=96100000from=StationList;freq=96000000from=StationList;freq=95900000from=StationList;freq=95800000from=StationList;freq=98700000from=StationList;freq=98500000from=StationList;freq=98600000from=StationList;freq=98700000from=StationList;freq=98800000from=StationList;freq=98900000from=StationList;freq=99000000from=StationList;freq=99100000
```

PE5PVB, 2023-08-04:
OK. Looks like XDRGTK is not sending a RDSclear command. I removed this from my build as well, can you check if it works now?

andimik, 2023-08-04:
This was also my idea, but that did not help.

Tested with a0e63c701d.

BTW: Please don't use the main branch for testing ...

PE5PVB, 2023-08-04:
OK. I did some more optimising. I will release a beta with todays changes. Maybe it works now.

andimik, 2023-08-04:
Will try in the evening.

andimik, 2023-08-05:
Unfortunately, it did not work and I guess I know the reason. 

**TEF_tuner does not transmit the frequency, only Bandwidth and RcvLevel**!

Seems the lines
https://github.com/PE5PVB/TEF6686_ESP32/blob/dac2b6fadf95ec6c8d46775bd23e90328682e378/TEF6686_ESP32.ino#L2903-L2906 
are not transmitted at all! Please see `Bandwidth` in capital letters, which comes from https://github.com/PE5PVB/TEF6686_ESP32/blob/dac2b6fadf95ec6c8d46775bd23e90328682e378/TEF6686_ESP32.ino#L3131

And RcvLevel comes from https://github.com/PE5PVB/TEF6686_ESP32/blob/dac2b6fadf95ec6c8d46775bd23e90328682e378/TEF6686_ESP32.ino#L2980

I've installed a pcap app in Android and made the following captures.

# Direct via TEF and Stationlist for Android

```
from=StationList;freq=?
from=TEF_tuner;RcvLevel=17
from=StationList;freq=95900000
from=StationList;freq=96500000
from=StationList;freq=97700000
from=StationList;freq=98400000
from=StationList;freq=99000000
from=TEF_tuner;RcvLevel=17
from=StationList;freq=99700000
from=StationList;freq=100300000
from=StationList;freq=100500000
from=TEF_tuner;RcvLevel=19
from=TEF_tuner;RcvLevel=19
from=TEF_tuner;RcvLevel=19
from=TEF_tuner;RcvLevel=19
from=TEF_tuner;Bandwidth=56000
from=TEF_tuner;RcvLevel=20
from=StationList;freq=100400000
from=StationList;freq=100200000
from=StationList;freq=99700000
from=StationList;freq=99300000
from=StationList;freq=98900000
from=StationList;freq=98400000
from=StationList;freq=98000000
from=StationList;freq=97900000
from=TEF_tuner;RcvLevel=46
from=TEF_tuner;RcvLevel=43
from=TEF_tuner;RcvLevel=40
from=TEF_tuner;Bandwidth=114000
from=TEF_tuner;RcvLevel=36
from=TEF_tuner;RcvLevel=35
from=StationList;freq=98100000
from=StationList;freq=98400000
from=StationList;freq=99300000
from=StationList;freq=99700000
from=StationList;freq=100100000
from=StationList;freq=100400000
from=StationList;freq=100600000
from=StationList;freq=100700000
from=StationList;freq=100800000
from=TEF_tuner;RcvLevel=20
from=TEF_tuner;RcvLevel=20
from=TEF_tuner;Bandwidth=114000
from=TEF_tuner;RcvLevel=20
from=TEF_tuner;RcvLevel=20
from=TEF_tuner;RcvLevel=23
from=TEF_tuner;RcvLevel=23
from=StationList;freq=100700000
from=StationList;freq=100500000
from=StationList;freq=99600000
from=StationList;freq=98400000
from=TEF_tuner;RcvLevel=11
from=StationList;freq=98000000
from=StationList;freq=97700000
from=StationList;freq=97500000
from=TEF_tuner;RcvLevel=12
from=TEF_tuner;RcvLevel=13
from=TEF_tuner;RcvLevel=13
from=TEF_tuner;RcvLevel=14
from=TEF_tuner;Bandwidth=114000
from=TEF_tuner;RcvLevel=15
from=StationList;freq=97600000
from=TEF_tuner;RcvLevel=24
from=StationList;freq=97700000
from=TEF_tuner;RcvLevel=22
from=StationList;freq=97800000
from=TEF_tuner;RcvLevel=22
from=StationList;freq=97900000
from=TEF_tuner;RcvLevel=44
from=StationList;freq=98000000
from=TEF_tuner;RcvLevel=36
from=StationList;freq=98100000
from=TEF_tuner;RcvLevel=30
```
(I've made a new line here in Github)

[PCAPdroid_05_Aug._18_.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12267761/PCAPdroid_05_Aug._18_.zip)
[laptop_xdrgtk.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12267762/laptop_xdrgtk.zip)
[PCAPdroid_05_Aug._18_38_43.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12267763/PCAPdroid_05_Aug._18_38_43.zip)

[PCAPdroid_05_Aug._19_27_39.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12267777/PCAPdroid_05_Aug._19_27_39.zip)

[PCAPdroid_05_Aug._19_48_51.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12267801/PCAPdroid_05_Aug._19_48_51.zip)

Well, I've changed the lines to 

```
  if (wifi) {
    Udp.beginPacket(remoteip, 9030);
    Udp.print("from=TEF_tuner;RcvLevel=");
    Udp.print(String(SStatus / 10));
    Udp.print("from=TEF_tuner;freq=");
      if (band > BAND_GAP) {
        Udp.print(String(frequency_AM) + "000");
      } else {
        Udp.print(String(band == BAND_FM ? frequency : frequency_OIRT) + "0000");
      }
    Udp.endPacket();
  }
```

So at least the frequency is transmitted, but it seems that also other parameters are needed for *Stationlist for Android*

```

from=StationList;freq=?
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;freq=100100000
from=StationList;freq=?
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;freq=100100000
from=StationList;freq=100200000
from=StationList;freq=100400000
from=StationList;freq=101400000
from=StationList;freq=101900000
from=StationList;freq=102300000
from=StationList;freq=102700000
from=StationList;freq=102900000
from=StationList;freq=103100000
from=StationList;freq=103200000
from=StationList;freq=103300000
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;freq=100100000
from=TEF_tuner;Bandwidth=64000
from=TEF_tuner;RcvLevel=46
from=TEF_tuner;freq=100400000
from=TEF_tuner;RcvLevel=43
from=TEF_tuner;freq=101400000
from=TEF_tuner;RcvLevel=42
from=TEF_tuner;freq=101900000
from=TEF_tuner;RcvLevel=40
from=TEF_tuner;freq=102300000
from=StationList;freq=103400000
from=TEF_tuner;RcvLevel=51
from=TEF_tuner;freq=104000000
from=StationList;freq=103500000
from=TEF_tuner;RcvLevel=15
from=TEF_tuner;freq=103400000
from=StationList;freq=103000000
from=TEF_tuner;RcvLevel=46
from=TEF_tuner;freq=104000000
from=StationList;freq=102500000
from=TEF_tuner;RcvLevel=21
from=TEF_tuner;freq=103000000
from=StationList;freq=102000000
from=TEF_tuner;RcvLevel=24
from=TEF_tuner;freq=102500000
from=StationList;freq=101500000
from=TEF_tuner;RcvLevel=15
from=TEF_tuner;freq=102000000
```

[PCAPdroid_05_Aug._20_07_29.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12267879/PCAPdroid_05_Aug._20_07_29.zip)

PE5PVB, 2023-08-06:
Frequency is send when frequency is changed. This works fine with Stationlist on PC.

andimik, 2023-08-06:
Indeed. I have captured that when connected to Stationlist for Windows (running under Wine and Ubuntu).

[tef_stationlist_windows_short.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12269941/tef_stationlist_windows_short.zip)
[tef_stationlist_windows.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12269942/tef_stationlist_windows.zip)

```
from=StationList;freq=102900000
```

```

from=TEF_tuner;RT1=4d555241545f265f4a4f53455f464541542e5f4d4159415f2d5f504f5354414e5f54495f53414d008000000039a73900c0c1403f40bd0f40f4aa0d8040dbfc3f
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;RT1=4d555241545f265f4a4f53455f464541542e5f4d4159415f2d5f504f5354414e5f54495f53414d001448fc3fcc46fc3fc6250000000000fff4aa0d8040dbfc3f
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;RT1=4d555241545f265f4a4f53455f464541542e5f4d4159415f2d5f504f5354414e5f54495f53414d008000000039a73900c0c1403f40bd0f40f4aa0d8040dbfc3f
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;RT1=4d555241545f265f4a4f53455f464541542e5f4d4159415f2d5f504f5354414e5f54495f53414d008000000039a73900c0c1403f40bd0f40f4aa0d8040dbfc3f
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;RcvLevel=52;bandwidth=-1;freq=102900000
from=TEF_tuner;PTY=20
from=TEF_tuner;RT1=00000000fffffffff4aa0d8040dbfc3f9c58fc3fb7d4a2145f210000100000001a0000000100008093d80000349f4a3fc0a82bfc000000000000000000000000
from=TEF_tuner;PI=....
from=TEF_tuner;PS=0000000000c0a82b
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;RT1=00c2403f40bd0f40fb9d0d805fdbfc3f3457fc3f0100000038000000040000001a000000000000800050f63f050000002300000016000000f4aa0d8040dbfc3f
from=TEF_tuner;RcvLevel=52
from=TEF_tuner;RT1=007efb3f00000000fb9d0d805fdbfc3f3457fc3f0100000038000000040000001a00000001000080b7d4a2146461fc3f744bfc3f764bfc3ff4aa0d8040dbfc3f
from=TEF_tuner;RcvLevel=53
from=TEF_tuner;RT1=007efb3f00000000fb9d0d805fdbfc3f3457fc3f0100000038000000040000001a000000010000808000000039a73900c0c1403f40bd0f40f4aa0d8040dbfc3f
from=TEF_tuner;RcvLevel=53
from=TEF_tuner;RT1=007efb3f00000000fb9d0d805fdbfc3f3457fc3f0100000038000000040000001a000000000000808000000039a73900c0c1403f40bd0f40f4aa0d8040dbfc3f
from=TEF_tuner;RcvLevel=53
from=TEF_tuner;RT1=00000000f8000000fb9d0d805fdbfc3f3457fc3f0100000038000000040000001a00000001000080b7d4a2146461fc3f000000008cdafc3ff4aa0d8040dbfc3f
from=TEF_tuner;PI=A213
from=TEF_tuner;RcvLevel=53
from=TEF_tuner;RT1=00000000a058fc3ffb9d0d805fdbfc3f3457fc3f0100000038000000040000001a00000001000080b7d4a2146461fc3f0100000060dafc3ff4aa0d8040dbfc3f
```

And then back on Android, no frequency, no AF

```
from=TEF_tuner;RcvLevel=36
from=TEF_tuner;RcvLevel=10
from=TEF_tuner;RcvLevel=13
from=TEF_tuner;RcvLevel=13
from=TEF_tuner;RcvLevel=35
from=TEF_tuner;RcvLevel=7
from=TEF_tuner;RcvLevel=8
from=TEF_tuner;RcvLevel=8
from=TEF_tuner;RcvLevel=8
from=TEF_tuner;RcvLevel=8
from=TEF_tuner;RcvLevel=8
from=TEF_tuner;Bandwidth=64000
from=TEF_tuner;RcvLevel=8
from=TEF_tuner;RcvLevel=8
from=TEF_tuner;RcvLevel=8
from=TEF_tuner;RcvLevel=9
```

**Sjef, there is a bug indeed. If connected to *Stationlist for Android* another script from the sources are used!** See above, I already told you the lines.

Please see the word `bandwidth` resp. `Bandwidth`, it's not the same.

# TEF sends to Stationlist for Windows 

`from=TEF_tuner;RcvLevel=52;bandwidth=-1;freq=102900000` 

# TEF sends to Stationlist for Android

`from=TEF_tuner;Bandwidth=64000`

andimik, 2023-08-06:
Progress!

Now TEF transmits the frequency, but SL for Android still does not want to update to the new frequency.

```
from=StationList;freq=102900000
from=TEF_tuner;RcvLevel=59;bandwidth=-1;freq=99100000
from=TEF_tuner;PI=A3DD
from=TEF_tuner;PTY=3
from=TEF_tuner;PS=5355445449524f4c
from=StationList;freq=103000000
from=TEF_tuner;RcvLevel=56;bandwidth=-1;freq=102900000
from=TEF_tuner;PI=A213
from=TEF_tuner;PTY=0
from=TEF_tuner;PS=5f5f464d345f5f5f
from=StationList;freq=103100000
from=TEF_tuner;RcvLevel=25;bandwidth=-1;freq=103000000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
from=StationList;freq=103200000
from=TEF_tuner;RcvLevel=13;bandwidth=-1;freq=103100000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
from=StationList;freq=103300000
from=TEF_tuner;RcvLevel=8;bandwidth=-1;freq=103200000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
from=StationList;freq=103400000
from=TEF_tuner;RcvLevel=9;bandwidth=-1;freq=103300000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
from=StationList;freq=105200000
from=TEF_tuner;RcvLevel=48;bandwidth=-1;freq=105500000
from=TEF_tuner;PI=A550
from=TEF_tuner;PTY=0
from=TEF_tuner;PS=5f5f41474f52415ffrom=StationList;freq=104000000
from=TEF_tuner;RcvLevel=23;bandwidth=-1;freq=105200000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
from=StationList;freq=102500000
from=TEF_tuner;RcvLevel=52;bandwidth=-1;freq=104000000
from=TEF_tuner;PI=9202
from=TEF_tuner;PTY=9
from=TEF_tuner;PS=56414c5f3230325f
from=StationList;freq=105500000
from=TEF_tuner;RcvLevel=25;bandwidth=-1;freq=102500000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
```

https://github.com/PE5PVB/TEF6686_ESP32/blob/dac2b6fadf95ec6c8d46775bd23e90328682e378/TEF6686_ESP32.ino#L2978-L2983 and following should be

```
  if (wifi) {    // for Stationlist for Android modified by andimik
    Udp.beginPacket(remoteip, 9030);
    Udp.print("\nfrom=TEF_tuner;RcvLevel=");
    Udp.print(String(SStatus / 10));
    Udp.print(";bandwidth=-1;freq=");
      if (band > BAND_GAP) {
        Udp.print(String(frequency_AM) + "000");
      } else {
        Udp.print(String(band == BAND_FM ? frequency : frequency_OIRT) + "0000");
      }
    Udp.print("\nfrom=TEF_tuner;PI=" + String(radio.rds.picode, 4));
    Udp.print("\nfrom=TEF_tuner;PTY=");
    Udp.print(String(radio.rds.stationTypeCode, HEX));
          Udp.print("\nfrom=TEF_tuner;PS=");
      char PShex[9];
      radio.rds.stationName.toCharArray(PShex, 9);
      for (int i = 0; i < 8; i++)
      {
        if (PShex[i] < 0x10) Udp.print("0");
        if (PShex[i] == 0x20) PShex[i] =  '_';
        Udp.print(String(PShex[i], HEX));
      }
    Udp.endPacket();
  }
```

But then we have the same code again in the source code ...

PE5PVB, 2023-08-06:
So with a leading \n it works?

andimik, 2023-08-06:
No, I tried that just for better formatting, but no influence.

Well, it does not work, but the texts are being transmitted.

```

from=StationList;freq=105700000
from=TEF_tuner;RcvLevel=20;freq=88300000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
from=StationList;freq=105800000
from=TEF_tuner;RcvLevel=6;freq=105700000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
from=StationList;freq=105900000
from=TEF_tuner;RcvLevel=5;freq=105800000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
from=StationList;freq=106000000
from=TEF_tuner;RcvLevel=16;freq=105900000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
from=StationList;freq=104900000
from=TEF_tuner;RcvLevel=36;freq=87600000
from=TEF_tuner;PI=51CC
from=TEF_tuner;PTY=7
from=TEF_tuner;PS=522e4d415249415f
from=StationList;freq=105500000
from=TEF_tuner;RcvLevel=57;freq=104900000
from=TEF_tuner;PI=A540
from=TEF_tuner;PTY=a
from=TEF_tuner;PS=414e54454e4e455f
```

Maybe I need to investigate, why the IP-Address is different.

[image]

I am expecting my mobile phone as 192.168.43.1, not as 10.215.173.1.

andimik, 2023-08-07:
testing 77e8596

Do you expect anything new in 2495ff as you have just changed from uppercase `Udp.print("from=TEF_tuner;Bandwidth=");` to lowercase `Udp.print("from=TEF_tuner;bandwidth=");`? 

This has no influence on Stationlist for Android.

```
from=StationList;freq=105900000
from=TEF_tuner;RcvLevel=58
from=StationList;freq=106000000
from=TEF_tuner;RcvLevel=16
from=StationList;freq=104000000
from=TEF_tuner;RcvLevel=38
from=StationList;freq=105500000
from=TEF_tuner;RcvLevel=51
from=StationList;freq=104500000
from=StationList;freq=104500000
from=StationList;freq=104900000
from=StationList;freq=104000000
from=StationList;freq=106700000
from=TEF_tuner;RcvLevel=42
from=TEF_tuner;RcvLevel=43
from=TEF_tuner;RcvLevel=44
from=TEF_tuner;RcvLevel=45
from=TEF_tuner;RcvLevel=45
from=StationList;freq=106800000
from=TEF_tuner;RcvLevel=23
from=StationList;freq=105200000
from=TEF_tuner;RcvLevel=15
from=StationList;freq=105300000
from=TEF_tuner;RcvLevel=22
```

In order to achieve better communication with SL for Android, you need to modify the other code mentioned above, which is now in L2342ff (= inside of `void ShowSignalLevel()`)

PE5PVB, 2023-08-07:
Can you make an example what you want to happen? It's just plain UDP traffic.

andimik, 2023-08-07:
Yes, I am aware what UDP traffic is, but there are two (2) different codes for UDP traffic at the moment, which I find quite useless.

One is for **SL for Windows** (which you have shifted to `rds.cpp`, lines 650, 663, 683, 745), the other is for **SL for Android** (still in the ino-file)

See above, currently there is **no** frequency information transmitted over UDP for **SL for Android**.

In order to update the frequency on **SL for Android** TEF should transmit at least the frequency, but it doesn't.

Please see my patch above, it will transmit the correct info even for **SL for Android**, but I still need to test the communication.

PE5PVB, 2023-08-07:
Well, there is no detection of any connection, so the UDP packets are always transmitted when WiFi is connected, even when you don't run stationlist on any device at all.

andimik, 2023-08-07:
Are you sure?

PE5PVB, 2023-08-07:
Yup. Try it.

PE5PVB, 2023-08-07:
When you change frequency it will send UDP packets. Every time when a new frequency is shown on the display and wifi is enabled it will run this:
    if (wifi) {
      String stationprint;
      stationprint = "from=TEF_tuner;RcvLevel=";
      stationprint += String(SStatus / 10);
      stationprint += ";bandwidth=-1;freq=";
      if (band > BAND_GAP) {
        stationprint += String(frequency_AM) + "000";
      } else {
        stationprint += String(band == BAND_FM ? frequency : frequency_OIRT) + "0000";
      }
      Udp.beginPacket(remoteip, 9030);
      Udp.print(stationprint);
      Udp.endPacket();
    }

PE5PVB, 2023-08-07:
Make sure your Stationlist is listening to port 9030

andimik, 2023-08-07:
> Make sure your Stationlist is listening to port 9030

Of course it does. I cannot change that in the tool ("SL listens to Port -1"), it's hardcoded.

I tried several times, when I connect my mobile I get other results than when I connect my PC.

Modifying L2352 with 

```
  if (wifi) {
 // for Stationlist for Android modified by andimik
    Udp.beginPacket(remoteip, 9030);
    Udp.print("\nfrom=TEF_tuner;RcvLevel=");
    Udp.print(String(SStatus / 10));
    Udp.print(";bandwidth=-1;freq=");
      if (band > BAND_GAP) {
        Udp.print(String(frequency_AM) + "000");
      } else {
        Udp.print(String(band == BAND_FM ? frequency : frequency_OIRT) + "0000");
      }
    Udp.print("\nfrom=TEF_tuner;PI=" + String(radio.rds.picode, 4));
    Udp.print("\nfrom=TEF_tuner;PTY=");
    Udp.print(String(radio.rds.stationTypeCode, HEX));
          Udp.print("\nfrom=TEF_tuner;PS=");
      char PShex[9];
      radio.rds.stationName.toCharArray(PShex, 9);
      for (int i = 0; i < 8; i++)
      {
        if (PShex[i] < 0x10) Udp.print("0");
        if (PShex[i] == 0x20) PShex[i] =  '_';
        Udp.print(String(PShex[i], HEX));
      }
    Udp.endPacket();
  }
```

will transmit the frequency!!!

> Every time when a new frequency is shown on the display and wifi is enabled it will run this:

No it does not!

PE5PVB, 2023-08-07:
Please tell me what your script does. I'm a bit confused now.

andimik, 2023-08-07:
It's **your** script just copied to another location.

andimik, 2023-08-07:
forget `\n`, this is just cosmetic.

PE5PVB, 2023-08-07:
But how are you sure it's not transmitted to your phone? How do you log this?

andimik, 2023-08-07:
Please see above, I've recorded some pcap files for that reason, the capture is done on the **same** device (mobile resp. laptop), where Stationlist is running (Android resp. Windows).

The UDP packets are **not** the same for SL for Windows and SL for Android.

I've realized that because you made a different spelling in the word "bandwidth" by chance.

Captured with modified code from above on the **right** place in the source code.

[PCAPdroid_07_Aug._23_38_58.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12285441/PCAPdroid_07_Aug._23_38_58.zip)

```
from=StationList;freq=104800000
from=TEF_tuner;RcvLevel=44;bandwidth=-1;freq=105500000
from=TEF_tuner;PI=A550
from=TEF_tuner;PTY=0
from=TEF_tuner;PS=5f5f41474f52415ffrom=StationList;freq=104900000
from=TEF_tuner;RcvLevel=22;bandwidth=-1;freq=104800000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000from=StationList;freq=?
from=TEF_tuner;RcvLevel=53;bandwidth=-1;freq=106000000
from=TEF_tuner;PI=9203
from=TEF_tuner;PTY=0
from=TEF_tuner;PS=5f5f4152535f5f5ffrom=StationList;freq=105000000
from=TEF_tuner;RcvLevel=53;bandwidth=-1;freq=106000000
from=TEF_tuner;PI=9203
from=TEF_tuner;PTY=0
from=TEF_tuner;PS=5f5f4152535f5f5ffrom=StationList;freq=105100000
from=TEF_tuner;RcvLevel=13;bandwidth=-1;freq=105000000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000from=StationList;freq=105200000
from=TEF_tuner;RcvLevel=7;bandwidth=-1;freq=105100000
from=TEF_tuner;PI=....
from=TEF_tuner;PTY=20
from=TEF_tuner;PS=0000000000000000
```

Without this modification you will never see a frequency from TEF_tuner.

andimik, 2023-08-08:
In the meantime I had contact with the author of Stationlist. He gave me a hint, and I will test that in the next days.

So no further need to investigate here at the moment.

andimik, 2023-08-14:
I'll give up, sorry. There is definitely a bug, but I cannot tell you if the problem is on **TEF6686** side or even on **Stationlist-M** side.

Could you test it on your mobile please?

I've tried my mobile and an Android tablet, same result.

Fact is: TEF will NOT transmit the frequency when you change the frequency on TEF.

Please do **not** test it in Stationlist for Windows. Here it works.

> When you change frequency it will send UDP packets. Every time when a new frequency is shown on the display and wifi is enabled it will run this:

Are you sure? I do **not** see that. During this capture, I was changing the frequency on the TEF quite often. There is nothing in the pcap-file.

```
from=StationList;freq=87600000
from=TEF_tuner;RcvLevel=33
from=StationList;freq=87700000
from=TEF_tuner;RcvLevel=37
from=StationList;freq=87800000
from=TEF_tuner;RcvLevel=19
from=StationList;freq=87900000
from=TEF_tuner;RcvLevel=13
from=StationList;freq=?
from=TEF_tuner;RcvLevel=47
from=StationList;freq=?
from=TEF_tuner;RcvLevel=47
```

[PCAPdroid_14_Aug._12_17_07.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12333848/PCAPdroid_14_Aug._12_17_07.zip)

PE5PVB, 2023-08-14:
I checked again, the only time that it was not transmitted over UDP was when screenmute was on. I fixed that, even when I now set the destination address to my doorbell, UDP packets are send.

PE5PVB, 2023-08-14:
No, but again, I checked. Now it always transmits the UDP packet to the destination device. I tested it with portmapping and send it to my doorbell, my router and my kids toy. So if your phone is still not receiving it, it means your phone's firewall is filtering it. I have to make a few small changes and upload the new version.

andimik, 2023-08-15:
Remark:

https://github.com/kkonradpl/xdr-gtk/blob/616d5015dc1f49aec76d4fc70e654112b7f20378/src/conf-defaults.h#L87-L97

`xdr-gtk` **only** uses the port, but **no** IP address can be set in the program. And it **works** with *SL-M*. 

The author of *Stationlist* gave me the hint that *TEF* should answer to **that** IP address where the question came from.

He told that TEF might use a **wrong** IP address as my pcap files do not show all the UDP packets which we expect.

In the meantime I have tested three Android devices. Same result.

The only thing I could try is to use another network (with a WiFi router). But then you can only use it at home and you won't need the Android version ...

Because up to now the Android mobile is creating a hotspot (192.168.43.1).

mx4zxc, 2023-08-15:
It works on my phone when I set the phone's ip address in the tef6686 radio. And I changed Stationlist ID 1.

andimik, 2023-08-15:
@mx4zxc

When you change the frequency on the TEF, will it change on Stationlist-M as well?

Do you use Hotspot or another network?

mx4zxc, 2023-08-15:
Yes, it works well both in the home network and when I use the phone as a hotspot. But there is a small bug. When I tune with tef and switch to tuning with the app, the app does not update when I tune again with tef. Then I have to go into the settings and change the station list ID to another number... after that the app updates when I tune with tef again. Attached are some pictures of my setup.
[image]
[image]

mx4zxc, 2023-08-15:
I really hope you get around to this, because it's a very useful tool. Thank you very much for sharing the link and information with me.

andimik, 2023-08-15:
> But there is a small bug. When I tune with tef and switch to tuning with the app, the app does not update when I tune again with tef.

This is exactly the same bug I'm talking about all the time in this issue.

MANY THANKS for confirming that, too.

mx4zxc, 2023-08-15:
I'm so sorry, I misunderstood your problem. I thought you meant that tef never updated the correct frequency to the app when tuning with tef. Which it actually does.

andimik, 2023-08-15:
The next days I will test hardcoding the IP address, if this works, I will send the value of `remoteip` to the mobile and check if it is NULL or even a wrong IP.

mx4zxc, 2023-08-15:
@andimik It is good. Hope you figure it out. Anyway, I'm very glad that you made me aware of this app. It will be used a lot in the future :)

andimik, 2023-08-15:
And I will try `socat` on the terminal of my (not rooted mobile)

https://android.stackexchange.com/questions/168422/how-to-port-forward-internet-using-android-hotspot-to-pc

mx4zxc, 2023-08-15:
I have put the phone in hotspot mode and connected my tef6686 to the phone via WIFI.
1. Open the app JB's StationList-M(SL-M).
2. Change frequency with tef6686 and the new frequency is updated in the SL-M app.
3. Change the frequency in SL-M and the tef6686 automatically changes the frequency to the one I set in the SL-M app.
4. Change the frequency again with the tef6686, but this time the SL-M app is not updated with the new frequency that I set in the tef6686.
5. Hold down the mode button to enter the menu, then immediately press mode and exit the menu.
6. Change frequency with the tef6686 radio and the SL-M app is updated with the new frequency.

It seems that there is a two-way connection, but that the tef6686 does not update automatically after you change the frequency with an external device.
But it's enough to go into the menu and out again, then it starts sending the frequency change again.

Just tested holding the band button and it started sending frequency changes to the SL-M app again.

PE5PVB, 2023-08-15:
TEF writes data to port 9030 and listens to 9031. This is how the PC version of Stationlist works.

mx4zxc, 2023-08-16:
Yes, it seems very nice. Here I use the phone as a mobile Wi-Fi zone (hotspot):
https://youtu.be/Rzp4MHQUpw8

andimik, 2023-08-16:
Interesting. It works in your case. Which firmware and date (resp. commit) is that?

According to the new frequency font it is the latest, right?

andimik, 2023-08-16:
That's the way I am doing as well of course.

In the next days (when Sjef is on vacation) I will try to investigate why I only get fragments on the phone.

andimik, 2023-08-17:
In the meantime I found a device where it worked **until** I changed the frequency in SL-M. Then I had the same problem again and SL-M was not updated when TEF tunes its frequency. But the name was `TEF_tuner` all the time.

Then I changed the device and the name was `TEF_tuner 1` and it didn't work
Maybe it has to do if you tune to a RDS frequency or not.

As there is another bug in RDS I will send a Pull request ananyw

@mx4zxc

Can you please tell me, what is shown in Stationlist-M server name? `TEF_tuner 1` or just `TEF_tuner` (without space and number).

mx4zxc, 2023-08-17:
@andimik 
It doesn't matter which ID I choose. It works anyway.
[image]
In this part:
"In the meantime I found a device where it worked until I changed the frequency in SL-M. Then I had the same problem again and SL-M was not updated when TEF tunes its frequency."
Did you just try to go in to Advanced RDS and back?  
I have this problem too, but it is always solved by going into menu and back out, or by going into advanced rds and back.
I demonstrate this in the video from around 0:36

I helped a friend set it up yesterday and he has an older version of V2.0 beta that is 3-4 weeks old. And everything seemed fine with him as well.

andimik, 2023-08-17:
Oh, thank you for the hint with the advanced RDS view. Will try it later at home.

mx4zxc, 2023-08-17:
Another thing, If I don't define the StationList ID in TEF6686, then it will only show TEF_tuner in the app.

Yes, the video was actually directed at another group where I was just going to demonstrate the app JB's StatonList-M.

Hope it works for you, because this app is a useful tool both with and without TEF :)

[13 more replies]

closed 2023-09-07 by andimik