PE5PVB/TEF6686_ESP32 #161: Improvements for RT

lxsxl, 2023-07-29:
1. Dynamic RT typically changes flag A/B in about 15 seconds (sometimes only about 5 seconds). Those times comply to IEC 62106-2. Actually one transition in the TEF lasts about 45 seconds (in main screen) or 23 seconds (in advanced RDS screen). So most of the time RT is switching before transition is done. IMHO it can be speeded up so that transition is completed in about 15 seconds. With the smoothed fonts it should still be easily readable.

2. In case of flag A/B change RT should be reset and start the new text even if the previous transition is not done.

3. In the main screen RT scrolls until the last character has vanished beyond the left margin. This should be equalized to the advanced mode: start RT again after the last character (and a bit of empty space) has become visible.

PE5PVB, 2023-07-29:
1. So you ask to speed up the ticker?
2. AB flag is already used to clear RT when toggled
3. It's very difficult to tune the timing of the ticker. This is because the characters don't all have the same width. Like a i or l is shorter than a m or W.

lxsxl, 2023-07-30:
1. Yes, that's right. Even at double or triple speed RT is still easy to read.
2. Yes, I know. Let me explain it with an example: In advanced mode only about 20 of the 64 characters can be seen at once. Now RT is scrolling, and characters 20 to 40 are seen. If A/B-flag switches and the length of the new text is only 20 characters, it will not be visible until the 64 characters have finished. But in the meantime there may have been the next A/B-flag switch. So my suggestion is to always restart with the first character if there is a A/B-flag switch.
3. Okay, I understand. So you should keep it running like it is.
4. Maybe a further idea: IEC describes the use of 0x0D to terminate the RT-line. If you would comply with this it would also save time which is actually spent for a lot of blanks.