PE5PVB/TEF6686_ESP32 #166: Direct RDSSpy connection

andimik, 2023-07-31:
https://github.com/PE5PVB/TEF6686_ESP32/blob/48358719e23634b774d909bb012079d8a15e7c00/TEF6686_ESP32.ino#L309

*RDS Spy* connection seems to be configured in this line using **port 7373**. Of course `RDS Spy`in menu is set for the USB (and wifi) connection.

But in the latest versions (maybe last 2-3 weeks) I could not establish a direct *RDS Spy* connection. I even have tested if some `serial.print` outputs are printed (like I have seen before), but there is nothing on 7373.

I found my postings in a German radio forum dated 15th and 16th June, where I informed my colleagues, that *RDS Spy* works well. I remember that I could set the frequency in *RDS Spy* and the *TEF* re-tuned to it.

https://www.rundfunkforum.de/viewtopic.php?p=1716663#p1716663
https://www.rundfunkforum.de/viewtopic.php?p=1716856#p1716856

Switching back to `xdr-gtk` option in menu, and connection works well with *xdr-gtk*.

PE5PVB, 2023-08-03:
I've seen this too. I have to search why it doesn't work anymore. BTW. You don't have to set the RDS Spy in the menu, over WiFi it just should work when your not connected with XDRGTK

andimik, 2023-08-04:
I just reinstalled the first ever 2.00 version from 17th June and RDSSpy works as expected.

```
$ git show
commit 71634398bc08e1b99f96826e42d9e9df672a4b03 (HEAD -> main)
Author: Sjef Verhoeven PE5PVB <pe5pvb@het-bar.net>
Date:   Sat Jun 17 22:05:19 2023 +0200

    Added Stationlist support

diff --git a/TEF6686_ESP32.ino b/TEF6686_ESP32.ino
index 1d8e61b..2f64eed 100644
--- a/TEF6686_ESP32.ino
+++ b/TEF6686_ESP32.ino
@@ -31,19 +31,24 @@
 //#define ARS       // uncomment for BGR type display (ARS version)
 
 #ifdef ARS
-#define VERSION         "v1.16ARS"
+#define VERSION         "v2.00ARS"^M
 #include "TFT_Colors.h"
 TFT_eSPI tft = TFT_eSPI(320, 240);
 #else
-#define VERSION         "v1.16"
+#define VERSION         "v2.00"^M
```

PE5PVB, 2023-08-04:
Fixed.

andimik, 2023-08-04:
Have you tested 6b4f35b0 ? It seems not to work.

andimik, 2023-08-05:
Issue is not solved, please re-open it.

# Actual behavior

```
104800*F*D*R
105400*F*D*R
106500*F*D*R
108000*F*D*R
107400*F*D*R
107400*F*D*R
```
is sent to 7373, but no reaction from TEF.

TEF sends 

```
a0
a0
a0
```

[rdsspy_tef.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12267920/rdsspy_tef.zip)

# Expected behaviour (from former commit)

```
104900*F*D*R
101100*F*D*R
```
sent to TEF, and received:

```
G:
9201001e0f535649

G:
9201001f42532020

G:
92012007696a736b
```

[rdsspy_tef_working.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12267956/rdsspy_tef_working.zip)

PE5PVB, 2023-08-07:
I did some more improvements on the routine, please check again and let me know.

andimik, 2023-08-07:
Yes, now it works (again), thanks!

[image]

```
G:
a540254365742059

G:
a540054f5288454e

G:
a540054852ae4b41

G:
a54025446f75204c

G:
a540054952c74552

G:
a540054a52924e54

G:
a54025456f766520
```

[image]
[tef_rdsspy_working.zip](https://github.com/PE5PVB/TEF6686_ESP32/files/12285098/tef_rdsspy_working.zip)

closed 2023-08-07 by andimik