PE5PVB/TEF6686_ESP32 #185: Problem with patch #163, Screensaver issues.

mx4zxc, 2023-08-03:
Hello,
I still have problems with the screensaver.
When the screen saver is off and I long press the band button, a 3 sec delayed screen saver is forced on. Even in fm mode and I press the band button to get more RDS info.

I deleted these lines:
       if (touchrotating) {
         if (poweroptions != LCD_OFF) poweroptions = LCD_OFF;
         if (!screensaverset) screensaverset = 1;
       }

And everything was fine again.

I use the gray china model with silver buttons and have no problems with either the screensaver or anything else after I deleted the #163 patch.

onmytime, could you please check it?

ohmytime, 2023-08-03:
Hello, 
Please show more details, what is the status of those menu items?
touch rotating, screensaver , power options,

I tested with touch rotating, screensaver , power options all OFF, it's no problems.

Once if you turned on touch rotating, and then turned off, screensaver will keep ON, but it still works correct to show RDS info on FM band.

mx4zxc, 2023-08-04:
Thank you for the reply.
Ok, I see now that as long as touch rotating is ON, the screensaver will always be forced to turn ON.

But why must there be a relationship between touch rotating and screensavers? Was it better that it was the way it was, when touch rotating has relation to screen OFF?
At least I have no visible conflicts with skipping patch #163 as far as I can see.

So before we close this issue, you mentioned earlier that it created conflicts with the screensaver before patch #163, is there something I'm not seeing?

Mike86-CZE, 2023-08-04:
When I set Power options to 1...% brightness or Deep sleep, Screen off is set after returning to the menu.

Mike86-CZE, 2023-08-07:
Yes.  and I see no reason why there should be a connection between these functions.  after all, the screen can also be turned off manually in the AM bands where it is most desirable.

ohmytime, 2023-08-07:
Yes, it makes user confuse, so I have delete this function. #222

closed 2023-08-07 by ohmytime

mx4zxc, 2023-08-07:
I don't know what the point of this feature was, but I think it's very nice to be able to just go on the AM band and just turn off the screen when I found a broadcast I wanted to listen to.. and still have the ability to tune to other channels without having to take a trip into the menu.

But thanks for disabling the feature. :)