PE5PVB/TEF6686_ESP32 #199: Squelch values always in dBuV

HyperDX, 2023-08-05:
I've noticed that if I change signal unit in the settings to dBf or dBm squelch values on the main screen are still in dBuV and don't switch to dBf or dBm

ohmytime, 2023-08-05:
The squelch values do not use unit conversions.
It has a separate scope.

andimik, 2023-08-05:
Will this be changed?

ohmytime, 2023-08-05:
I don't think there will be a change, it's just a range. 
For example, the value range is 0-100, which has nothing to do with the signal unit.

andimik, 2023-08-05:
In my case it's 100 values too, but starting with `-9` and the maximum is `91`. All above is marked as `ST`, what does that mean?

ohmytime, 2023-08-06:
You can check out the doSquelch function

```
void doSquelch() {
  if (XDRGTKUSB == false && XDRGTKTCP == false) {
    Squelch = analogRead(PIN_POT) / 4 - 100;
    if (Squelch > 920) Squelch = 920;
    if (showsquelch && !advancedRDS && !afscreen) {
      if (menu == false && Squelch != Squelchold) {
        if (Squelchold == -100) {
          if (Squelch != Squelchold) tftPrint(-1, myLanguage[language][33], 235, 145, BackgroundColor, BackgroundColor, FONT16);
        } else if (Squelchold == 920) {
          if (Squelch != Squelchold) tftPrint(-1, "ST", 235, 145, BackgroundColor, BackgroundColor, FONT16);
        } else {
          if (Squelch != Squelchold) tftPrint(-1, String(Squelchold / 10), 235, 145, BackgroundColor, BackgroundColor, FONT16);
        }
        if (Squelch == -100) {
          if (Squelch != Squelchold) tftPrint(-1, myLanguage[language][33], 235, 145, PrimaryColor, PrimaryColorSmooth, FONT16);
        } else if (Squelch == 920) {
          tftPrint(-1, "ST", 235, 145, PrimaryColor, PrimaryColorSmooth, FONT16);
        } else {
          if (Squelch != Squelchold) tftPrint(-1, String(Squelch / 10), 235, 145, PrimaryColor, PrimaryColorSmooth, FONT16);
        }
      }
      Squelchold = Squelch;
    }
  }
```

Read analog value from pin and convert to ADC value( -100 ~ 920 ), The actual value displayed is String(Squelch / 10)

When Squelch == -100, print OFF on screen, when Squelch == 920 print ST.
So the actual value of Squelch is -9 ~ 91

andimik, 2023-08-06:
What does `ST` stand for?

ohmytime, 2023-08-06:
> What does `ST` stand for?

Stereo

andimik, 2023-08-06:
I did not expect that term in the squelch field.

ohmytime, 2023-08-06:
> I did not expect that term in the squelch field.

Well, that's the original design.

PE5PVB, 2023-08-06:
OK. Squelch will now also show dBuV, dBf or dBm values

closed 2023-08-06 by PE5PVB