PE5PVB/TEF6686_ESP32 #200: White display (crash?)

andimik, 2023-08-05:
Sometimes on my original v1 TEF I have a white display and I need to reboot it.

I don't know if this is related to WiFi (as I'm currently testing a lot) of will show after a while.

[image]

ohmytime, 2023-08-06:
which firmware?

andimik, 2023-08-06:
Latest of course

ohmytime, 2023-08-06:
Can you give more details , under what conditions would cause reboot ?

andimik, 2023-08-06:
I don't know. The power supply is plugged in, WiFi is on. Screensaver is off.

May happen after several seconds or many minutes.
Have not seen that in previous versions.

Up to now no chance to check if something was printed on the serial port.

PE5PVB, 2023-08-06:
Do you use a seperate front pcb?

andimik, 2023-08-06:
No, it is the original, without modification.

PE5PVB, 2023-08-06:
Try add the 100n caps directly on the back of the rotary encoder.

andimik, 2023-08-06:
I see, but I won't modify the hardware by myself. Just wanted to report it.

PE5PVB, 2023-08-06:
We had this issue in the past, it was solved by adding 100n caps on CW and CCW pin directly on the rotary encoder.

andimik, 2023-08-07:
Can you post a picture resp. a schematic?

andimik, 2023-08-07:
But what you describe is a simple reboot. Mine is a crash and freeze.

Let's see where the modifications has to be soldered.

PE5PVB, 2023-08-07:
Correct. When this happens the display shows only white, no reboot. Problem was caused by interference on the 3v3 powerrail of the display

closed 2023-08-31 by PE5PVB