PE5PVB/TEF6686_ESP32 #203: RT+ first shows mylanguage[0]

andimik, 2023-08-06:
When a RT+ transmission is received for the first time (after a cold start) RT+ will display mylanguage[0] from the language file until mylanguage[112ff] are shown.

If you set the language to - for example - German, the text is "Deutsch" of course.

I guess you need to wait till RT+ items are received and not just check for 0x4BD7. 

[image]

Remark: I only have this very disturbed radio station using RT+ here.

PE5PVB, 2023-08-06:
That is strange. Can you check again in a next update?

lxsxl, 2023-08-08:
It is still the same in yesterdays beta:
[image]

For me it seems that it happens when the device is switched on and then quickly switched to advanced RDS-screen. It will take a moment until RT+ is detected and that is the moment where the language can be seen, sometimes only a part of a second, somtimes it seems to get stuck,

PE5PVB, 2023-08-09:
Ah ok. It seems that you received block 3A with the RT+ application init, but not on of the assigned RT+ blocks. I'll fix the problem.

closed 2023-08-09 by PE5PVB