PE5PVB/TEF6686_ESP32 #206: WiFi icon appears twice due to new battery icon

andimik, 2023-08-06:
@ohmytime 

Dear Kevin, the battery symbol breaks the layout ...

The WiFi icon appears a second time some pixels left of the current one (and laying over the word kHz), but it's not the same.
And sometimes it's blinking now (either logo or no logo)

[VID_20230806_163927.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/b76813b8-146b-43df-992e-c328323cfaf0)

[image]

Tested with 1dbcd7d50

ohmytime, 2023-08-07:
When WiFI is on, battery detection is down. Anyway, I will look into this.

andimik, 2023-08-07:
I was not talking about battery :-)

ohmytime, 2023-08-07:
> I was not talking about battery :-)

Oh, sorry, you mean the battery icon is too wide , and  occupies the area of wifi icon ?

andimik, 2023-08-07:
No, the WiFi icon position is wrong for updates. Please check horizontal position.

ohmytime, 2023-08-07:
[image]

Thank you for your remind,  I know this problems now.

MCelliotG, 2023-08-07:
There were multiple instances of WiFi in the code in different places, looks like I missed one when placing the icon in the new position. It's an easy fix, but I'm on holiday right now.

ohmytime, 2023-08-07:
> There were multiple instances of WiFi in the code in different places, looks like I missed one when placing the icon in the new position. It's an easy fix, but I'm on holiday right now.

It's okay, I will look into it.

ohmytime, 2023-08-07:
#220 Fixed.

closed 2023-08-07 by ohmytime