PE5PVB/TEF6686_ESP32 #211: AF "filler" is misinterpreted

lxsxl, 2023-08-06:
The AF "filler" is actually misinterpreted. "Filler" is not related to a specific AF and does NOT mean that the frequency is a low-power-station.
Correct is as follows: The "filler" is only used in AF-method A AND when the count of AF is even. As AFs are always transmitted in pairs and within those pairs the first "AF" is the number of AFs, a completely filled set consists of an odd number of AFs. If the number is even, the last AF-byte is filled by "205" (= filler). See also IEC 62106-2 / 7.5 Coding of AF, pdf-page 63 in the last rdsforum-compendium.

For the advanced RDS-screen and also for the AF/EON-screen the "filler"-signals can be removed. If you want to keep the information on a transmitted filler, then it should be placed after the last AF.

HyperDX, 2023-08-06:
Thank you for this explanation! I agree that "filler" can be removed :)

andimik, 2023-08-07:
IEC does not mention the word "filler"

> In special cases they are transmitted in descending order, if they belong to different
regions, or carry from time to time different programmes

PE5PVB, 2023-08-07:
OK. I will remove filler. It is only confusing for everyone.

closed 2023-08-07 by PE5PVB