PE5PVB/TEF6686_ESP32 #235: AF / regional variants

lxsxl, 2023-08-08:
If you still like to enhance the AF function you could implement the regional version (only if AF method B is used). This works as follows:

In AF method B for each frequency of the network a set of AFs is transmitted. A single AF-set consists of the header (= number of frequencies + lead-frequency) and a few AF-pairs, where the lead-frequency is always one part of the pair.

If the AF is the same regional version as the lead-frequency, the pair is transmitted in ascending order (e. g. lead-frequency 100,4 -> AF-pair 100,4 - 100,8 or 98,6 - 100,4),
If the AF is a regional variant to the lead-frequency, the pair is transmitted in descending order (e. g. lead-frequency 100,4 -> AF-pair 100,4 - 91,8 or 101,0 - 100,4).

I will add a screenshot of RDS Spy, this may be more illustrative. "RV" means regional variant, might also be useful for the TEF.

Be careful: One frequency can be for the same RV and also for a different one (in RDS Spy 91,8 MHz), so this would have to be shown twice.

[image]

lxsxl, 2023-08-08:
And som more examples
[image]
 from IEC:

PE5PVB, 2023-08-08:
Can you get me a MPX recording on a station with AF method B?

lxsxl, 2023-08-08:
So you still have the one from WDR2? This should work fine. Otherwise I will record tomorrow in the evening.

PE5PVB, 2023-08-13:
I've added a detector for the regional flag. I can't test it here, since no stations are using it. Please test and report.

lxsxl, 2023-08-13:
Are the "regional" AFs the blue coloured? (If yes it doesn't match actually)

PE5PVB, 2023-08-13:
I used this lines, where buffer0 is f1 and buffer1 is f2:
                    if (buffer0 == currentfreq && buffer0 < buffer1) af[af_counter].regional = true;
                    if (buffer1 == currentfreq && buffer0 > buffer1) af[af_counter].regional = true;

PE5PVB, 2023-08-13:
Ah, I think I found a mistake here. I've just updated it.

andimik, 2023-08-14:
But why has the "filler" been removed? RDS Spy uses the term "RV" for *regional variant* which is easier to understand.

PE5PVB, 2023-08-14:
I will replace the color for RV. Hopefully this will fit, otherwise I will do (r). About filler, as far as I know this is only a placeholder and does not have anything to do with the frequency. Filler is used when you have an even number of frequencies. This was misunderstood by me first and is corrected.

PE5PVB, 2023-08-14:
Do you have any idea how to detect if a station uses method A or B?

lxsxl, 2023-08-14:
It’s the twelveth bit in group 2 block 2 (the one between PTY and RT segment adress).

lxsxl, 2023-08-14:
Yes, that‘s better. So we will see both colour for reception status and RV for regional indicator. Otherwise you would need six colours.Actually I‘m too busy to test intensively, so please be patient a few days …

PE5PVB, 2023-08-14:
No, AF method B is not even, because there is a leading number of AF's. For example
04, 99.5
104.5, 99.5
99.5 100.1
93.1 99.5
91.9 99.5

The 04 is, 4 AF frequencies follows. So method B should always be uneven and there is no need for a filler. In method A it can be even and then you need a filler for the unused space.

PE5PVB, 2023-08-14:
> It’s the twelveth bit in group 2 block 2 (the one between PTY and RT segment adress)

I can't find this in the documentation. Can you show me where you found this?

lxsxl, 2023-08-14:
> No, AF method B is not even, because there is a leading number of AF's. For example 04, 99.5 104.5, 99.5 99.5 100.1 93.1 99.5 91.9 99.5
>
> The 04 is, 4 AF frequencies follows. So method B should [...]

That's exactly what I tried to explain. Sorry if was likely misunderstood...

lxsxl, 2023-08-14:
> > It’s the twelveth bit in group 2 block 2 (the one between PTY and RT segment adress)
> 
> I can't find this in the documentation. Can you show me where you found this?

You can find it on pdf-page 55 (group typ 2A and 2B):
[image]

PE5PVB, 2023-08-14:
Yes, this is the RT AB flag, but I mean a way to detect if a station uses AF method A or AF method B

lxsxl, 2023-08-15:
I think this should work.

PE5PVB, 2023-09-06:
I've added a script do detect this, please test.

lxsxl, 2023-09-07:
Actually it still doesn‘t work. There are no RV marked for any station, and I checked a lot that do have them. At least the incorrect RV-markings have gone.
Could you implement a flag that shows If you detected method A or B, because actually I can‘t even evaluate this. And as this is the first step it could be helpful.

andimik, 2023-09-07:
If you want to check if A or B is used, why not use RDSSPY? It will also help you to identify possible stations which are using that function.

And you could share either a 192 kHz 16bit mono mpx recording or at least a spy file with the RDS.

PE5PVB, 2023-09-07:
Can you get me a MPX recording for testing, including the frequency from the recording?

lxsxl, 2023-09-07:
Flag for method A/B could help to see If at least the correct method was detected by the TEF (which I compare with RDS Spy of course). German public broadcasters aImost completely use method B, so I don‘t have to compare.

lxsxl, 2023-09-07:
***@***.*** sent you MPX and Spy recordings

1 item, 206 MB in total ・ Expires on 14 September, 2023

Download link:
https://wetransfer.com/downloads/fc7e38ab9ac259a99d6352ba927fd63620230907195816/b893453af6b3434074bb87937e3ec97c20230907195945/423ca9?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid

1 item

AF method B examples.zip - 206 MB

[Get more out of WeTransfer, get Pro](https://wetransfer.com/pro?trk=TRN_TDL_01&utm_campaign=TRN_TDL_01&utm_medium=email&utm_source=sendgrid)

About WeTransfer: https://wetransfer.com/about
Help: https://wetransfer.zendesk.com/hc/en-us
Legal: https://wetransfer.com/legal/terms

To make sure our emails arrive, please add ***@***.*** to [your contacts](https://wetransfer.zendesk.com/hc/en-us/articles/204909429).

PE5PVB, 2023-09-10:
Found it. Please test again.

lxsxl, 2023-09-11:
Something has changed, for some AFs the "R" is visible now. But it is not reproducible: If you tune to the same frequency later the "R" may have gone. I suppose that it is due to fact that AF method B uses several AF-lists and the RV-status may differ in each list. So there is still something misinterpreted.

If you check my recordings ([https://we.tl/t-bokw2VdTBc](https://we.tl/t-bokw2VdTBc)) you can see how RDS Spy is working on AF method B: At the beginning the frequencies are added to method A-section. Step by step the single AF-lists are detected as method B and tranferred to the method B-section. I suppose that Spy first collects all AF-pairs for the list, checks if the head-frequency is in the leading pair (together with the lists AF-number) AND in each AF-pair, then it is safely method B. In that way each AF-list is checked and processed. It is important that you read the RV-status only from one single AF-list (the one that refers to the tuned frequency) because RV-status may be different in another list (i. e. for a different frequency). And also important: If the TEF switches to another AF, the AF-list has to be rebuild (then of course refering to the new tuned frequency, I don't know if this is actually already done).

Idk if such routines are possible to be done by the TEF? Maybe it is too much data-collecting and logics for it? Anyway, I will check deeper in the next days and try to find some more reasons.

PE5PVB, 2023-09-12:
What I currently do is this:
1. Wait for the AF index (the number of AF's to follow).
2. Count the times the current frequency is read from the AF A or AF B.
3. If number of times current frequency was there exceeds 2, then Method B flag is set high

Please note. When opening AF view and all AF frequencies are read, but the flag was not set, the page will not be refreshed automaticly. I have to add this. So currently, just tune into a station, wait a while and then open AF to see the regional idents.