PE5PVB/TEF6686_ESP32 #245: display interference at 106.7 mhz

Yurij20, 2023-08-13:
On the latest version, there is interference from the display at a frequency of 106.7 mhz.  On firmware versions 1.16 1.21 this problem is not observed.  Please pay attention.  Thank you

PE5PVB, 2023-08-13:
This was done during compiling here. It has to do with a too high setting in the tft_espi configuration. When you compile it yourself it will be gone.

closed 2023-08-13 by PE5PVB

Mike86-CZE, 2023-08-14:
> On the latest version, there is interference from the display at a frequency of 106.7 mhz. On firmware versions 1.16 1.21 this problem is not observed. Please pay attention. Thank you

It is strange.  For me, the interference moved to approximately 100 MHz +-400 kHz.  There is no problem at 106.7 MHz.