PE5PVB/TEF6686_ESP32 #254: Mute icon even shown during audio

andimik, 2023-08-19:
My blue square TEF has a SQL squelch knob. When I turn it, the Mute icon is shown, but audio is normal.

[VID_20230819_110141.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/e349d0e6-868b-4baa-89a5-e6bbd4227db1)

andimik, 2023-08-19:
It seems this is due to the fact that I have used MPX as output.

I was assuming the headphone output will get the MPX, but obviously the internal (mono) speaker plays a combined audio incl. MPX.

AaronJowitt, 2023-08-26:
In MPX output mode, it's highly desirable that squelch works the same as in L/R mode. Useful when feeding an external stereo decoder, or an FM transmitter (low power around the house, or small fm station translator tx).

PE5PVB, 2023-08-26:
This is not possible because dsp is bypassed.

AaronJowitt, 2023-08-26:
> This is not possible because dsp is bypassed.

ok, that's a pity.  

thank you Sjef.

PE5PVB, 2023-10-01:
Muting will now also work when MPX mode is selected.

closed 2023-10-01 by HyperDX

AaronJowitt, 2023-10-15:
> Muting will now also work when MPX mode is selected.

Nice. thank you!