PE5PVB/TEF6686_ESP32 #255: Slow rotation of squelch knob produces pixel errors

andimik, 2023-08-19:
[VID_20230819_113522.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/99336de0-8c51-4ec1-870a-fce22e0c93b5)
[VID_20230819_113326.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/a432b2cf-06c8-405e-b062-1223bc7f2a78)

My new blue/dark green square TEF has a squelch knob.

When I rotate it very fast, the display is correct.

When I rotate it slow, there are display errors.

No matter what language is used (in the video it's English resp. German)

ohmytime, 2023-08-19:
That's a problem, but I don't have the test equipment and Sjef is on vacation.

andimik, 2023-08-19:
I know of course

PE5PVB, 2023-10-01:
I tried to reproduce this problem, but failed. Is it still in the latest software?

andimik, 2023-10-06:
Need to check on one of my TEFs with squelch in the next days.

andimik, 2023-10-06:
Yes, still the case. Tested with today's commit.

[image]

PE5PVB, 2023-10-06:
Can you make a movie how to reproduce?

andimik, 2023-10-07:
There are already two videos in the first posting.

PE5PVB, 2023-10-09:
I changed the way squelch value is shown to a sprite. This should avoid these problems in the future.

andimik, 2023-10-09:
The highest value should be `OFF` and the lowest `ST`, right?

But now you sometimes (depending on the speed you are turning the know) **cannot** reach these edges, the knob might even stop at `91` (or even lower) resp. `-9` (or even higher) but never reaching `OFF` resp. `ST`.

PE5PVB, 2023-10-10:
I think you have a harware problem with the potentiometer. Can you please ask others with the same model radio on Discord if they can reproduce this issue?

closed 2023-10-10 by andimik