PE5PVB/TEF6686_ESP32 #262: Switching to OIRT in SL-M keeps TEF in FM band

andimik, 2023-08-26:
@ohmytime 

Dear Kevin,

The following steps lets TEF think, OIRT is still the (old) FM band. I guess it has to do that you have introduced the OIRT as new band (which is unknown in SL-M). SL-M might just have band edges, and the 30 kHz steps are hard-coded in the app.

1. connect to Stationlist-M
2. switch TEF to FM
3. change to OIRT on Stationlist-M
4. display will show 65.9 MHz, but FM
[image]

Then the tuning stop (**only** tuning **up**) **on TEF** is 100 kHz and the variable is still `band == BAND_FM` instead of `band == BAND_OIRT`. I could see it in my own **debug** version what is sent to UDP

```
14:15:06.953 -> remoteip: 192.168.43.105; line 2203 with freq/bw/rcv (FM) 87700 kHz sent
14:15:28.588 -> remoteip: 192.168.43.105; line 2207 with freq/bw/rcv (AM) sent
14:15:29.394 -> remoteip: 192.168.43.105; line 2207 with freq/bw/rcv (AM) sent
14:15:30.845 -> remoteip: 192.168.43.105; line 2207 with freq/bw/rcv (AM) sent
14:15:32.168 -> remoteip: 192.168.43.105; line 2205 with freq/bw/rcv (OIRT) sent
14:15:39.906 -> remoteip: 192.168.43.105; line 2203 with freq/bw/rcv (FM) 87700 kHz sent
14:15:39.971 -> remoteip: 192.168.43.105; line 2203 with freq/bw/rcv (FM) 65900 kHz sent
14:16:36.324 -> remoteip: 192.168.43.105; line 2203 with freq/bw/rcv (FM) 66000 kHz sent
14:16:36.357 -> remoteip: 192.168.43.105; line 2203 with freq/bw/rcv (FM) 66000 kHz sent
14:16:36.453 -> remoteip: 192.168.43.105; line 2203 with freq/bw/rcv (FM) 66100 kHz sent
```

Hint: When you tune with SL-M the steps are 30 kHz as expected.

Whenever you tune down with TEF, it will immediately go to 108,00 and start seeking to the next found frequency in FM.

Can we detect if the frequency is in the OIRT range and change the variable `band` then?

ohmytime, 2023-09-04:
There is no support for Stationlist-M yet, needs to be done.

PE5PVB, 2023-10-01:
Auto bandselector for Stationlist added. Please test.

PE5PVB, 2023-10-06:
Should be fixed now. Please test

andimik, 2023-10-07:
No still buggy, when I switch from OIRT to FM, it may happen that the frequency will not change to FM but stays in OIRT band.

[VID_20231007_114511.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/9f6a7e9b-ac57-471b-80ec-6ed119286fde)

I wonder why on OIRT the complete display refreshes when I tune with SL. On other bands this is not the case.

[VID_20231007_114405.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/d0bae832-c18a-4469-a0e7-86b163fc1896)

PE5PVB, 2023-10-08:
Should be fixed now.

andimik, 2023-10-09:
Not 100% ok, switching in SL between SW and OIRT does not work properly. You will see an OIRT frequency on the display, but the sound is that from the previous SW station. So it does not **tune** to OIRT.

PE5PVB, 2023-10-09:
Ah, right. Hope I fixed it now.

andimik, 2023-10-10:
Not completely. When you now switch to LW in SL, TEF will switch to SW or to MW instead. Tuning in SL (Android says LW) leads to alternating switching between SW and MW but without changing the frequency.

For example: 
SL 153 kHz -> TEF 1700 kHz
SL 162 kHz -> TEF 522 kHz
SL 171 kHz -> TEF 1700 kHz
SL 180 kHz -> TEF 522 kHz

No matter if you have switched to LW (SL will show LW) with TEF. Any tuning in SL in LW will change between 1700 and 522 kHz.

PE5PVB, 2023-10-10:
So, current problems are only on LW/MW/SW? OIRT and FM are both ok now?

andimik, 2023-10-10:
I just had few minutes to test, but I could not face problems on FM and OIRT. Bandwidths and frequencies are OK now there.

PE5PVB, 2023-10-10:
My mistake, something forgotten ;-)
I can't test it right now, because I have no Stationlist at my current location, but can you check again?

andimik, 2023-10-10:
Better, but not solved. I switched to MW (SL displays 531 kHz), but TEF tuned to 1700 kHz.

And I still faced the problem of alternating switching between LW (or MW) and SW.

SL 162 kHz -> TEF 162 kHz (correct)
tune up in SL to 171 kHz -> TEF SW 1700 kHz
tune up in SL to 180 kHz -> TEF LW 180 kHz
tune up in SL to 189 kHz -> TEF SW 1700 kHz
tune up in SL to 198 kHz -> TEF LW 198 kHz

(also the same in MW band)

I fear that is caused as the band edges are overlapping. Because 1700 kHz ist SW for TEF and MW for SL ... ?

See picture:

[image]

Or you change to MW on TEF and tune to 1701 kHz in MW

[image]

https://github.com/PE5PVB/TEF6686_ESP32/blob/a5181c24632b09bdae1583cde448f25c09305c2c/src/constants.h#L40-L42

What do you think?

PE5PVB, 2023-10-10:
Thanks. Please test the fix.

andimik, 2023-10-11:
Great! Now it works, I can switch through all bands and the frequencies and tuning are correct!

Many thanks!!

Closing as solved in https://github.com/PE5PVB/TEF6686_ESP32/commit/e7d893e4a31df5147990100cb0337649d8a8764d

closed 2023-10-11 by andimik