PE5PVB/TEF6686_ESP32 #265: LIC (Language identification code)

andimik, 2023-08-28:
Whereas **RDS Spy** shows the LIC (Language identification code) at the same time it is transmitted (see below), **TEF** will only show it after several seconds or even never.

It's in RDS group 1A says the spec, 

> In Group 1A, Variant 3, Block 3 the language identification code is allocated according to 
Figure J.1. When implemented, this variant shall be transmitted at least once every 2 s. 

```
23:00:19 5293 1160 00E0 0000
23:00:19 5293 1160 3015 0000
23:00:21 5293 1160 00E0 0000
23:00:21 5293 1160 3015 0000
23:00:23 5293 1160 00E0 0000
23:00:23 5293 1160 3015 0000
23:00:24 5293 1160 00E0 0000
23:00:24 5293 1160 3015 0000
23:00:26 5293 1160 00E0 0000
23:00:26 5293 1160 3015 0000
23:00:28 5293 1160 00E0 0000
```

According to `iec62106_2009.pdf` 0x15 is Italian, which is correct here.

[image]

Here is a 16bit mono wave file 192kHz (playable also in **RDS Spy** of course)

http://andimik.bplaced.net/temp/freccia_192khz_16bit.zip

# good example

[image]

# bad example

ECC is shown, but LIC is missing, even after several seconds waiting

[image]

PE5PVB, 2023-09-06:
What I do is this:
              if (rds.rdsC >> 12 == 3) {                                                          // LIC code readout
                rds.LIC = rds.rdsC & 0xff;
                rds.hasLIC = true;
              }
            }

So, when bit 12 and bit 13 of RDS block C are 1, it's indicated that LIC is used and the code can be read.
Then I read the 8 bits of Block C to get the LIC code. and Flag LIC is turned to true.
Language is translated from code to text, you can find this in language.h

This all is only done when error threshold is not hit.

This is the only way I think you can detect LIC.

andimik, 2023-09-08:
There is another bug with LIC.

When it was detected as "Unknown" it won't be cleared on frequency change.

In the Screenshot during a bandscan I recognized this entry, although the station does not transmit LIC at all.

# before reboot

[image]

# after reboot

[image]

PE5PVB, 2023-09-10:
Fixed that. Please test.

andimik, 2023-09-16:
No, there is still a bug in your code, sorry to say that.

LIC is either shown immediately after reception (very rare the case), not shown at all (some of the cases) or takes too long (lots of cases). 

I've repeated it several times (50?) with the same station with the same latest compiled firmware (= I spend nearly the whole day for that). Also compiled with some debug messages. But it showed the same delay like the display.

```
git diff
diff --git a/src/rds.cpp b/src/rds.cpp
index 489ace0..526fc4d 100644
--- a/src/rds.cpp
+++ b/src/rds.cpp
@@ -61,6 +61,10 @@ void ShowAdvancedRDS() {
         if (radio.rds.hasLIC) tftPrint(-1, myLanguages[radio.rds.LIC], 242, 208, PrimaryColor, PrimaryColorSmooth, 16); else tftPrint(-1, "N/A", 242, 208, PrimaryColor, PrimaryColorSmooth, 16);
       }
       LIColdString = myLanguages[radio.rds.LIC];
+      Serial.print("\nLIColdString is now ");
+      Serial.print(LIColdString);
+      Serial.print(" ; hasLIC is now ");
+      Serial.print(radio.rds.hasLIC);
     } else {
       byte lic;
       switch (radio.rds.LIC) {
@@ -82,6 +86,10 @@ void ShowAdvancedRDS() {
       LIColdString = myLanguages[lic];
     }
     licold = radio.rds.LIC;
+    Serial.print("\nlicold is now ");
+    Serial.print(licold);
+    Serial.print(" ; hasLIC is now ");
+    Serial.print(radio.rds.hasLIC);
   }
 
   String pinstring = String(radio.rds.pinDay) + " " + String(radio.rds.pinHour) + ":" + (radio.rds.pinMin < 10 ? "0" : "") + String(radio.rds.pinMin);
```

It has nothing to do with reception, see below. LIC is **not detected** as it should be every max. 2 seconds.
```
23:00:19 5293 1160 00E0 0000
23:00:19 5293 1160 3015 0000
23:00:21 5293 1160 00E0 0000
23:00:21 5293 1160 3015 0000
23:00:23 5293 1160 00E0 0000
23:00:23 5293 1160 3015 0000
23:00:24 5293 1160 00E0 0000
23:00:24 5293 1160 3015 0000
23:00:26 5293 1160 00E0 0000
23:00:26 5293 1160 3015 0000
23:00:28 5293 1160 00E0 0000
```

The information about LIC is very fast in *redsea* and *rdsspy*.

For that reason, I've connected TEF to *xdr-gtk* and checked RDS on port 7376 in *redsea*. LIC was read out as expected as Italian, see below, **whereas TEF still showed N/A**.

```
Sep 16 17:26:54 {"pi":"0x5293","group":"1A","bler":0,"has_linkage":false,"language":"Italian","prog_type":"Rock music","tp":false}
Sep 16 17:26:56 {"pi":"0x5293","group":"1A","bler":0,"has_linkage":false,"language":"Italian","prog_type":"Rock music","tp":false}
Sep 16 17:26:58 {"pi":"0x5293","group":"1A","bler":0,"has_linkage":false,"language":"Italian","prog_type":"Rock music","tp":false}
Sep 16 17:27:00 {"pi":"0x5293","group":"1A","bler":0,"has_linkage":false,"language":"Italian","prog_type":"Rock music","tp":false}
Sep 16 17:27:01 {"pi":"0x5293","group":"1A","bler":0,"has_linkage":false,"language":"Italian","prog_type":"Rock music","tp":false}
Sep 16 17:27:03 {"pi":"0x5293","group":"1A","bler":0,"has_linkage":false,"language":"Italian","prog_type":"Rock music","tp":false}
```

I agree you need to check group 1A, Block C, bits 12 and 13 for 1 each. See line 605ff. This will happen every max. 2 seconds according to the IEC standard. See log above.

I cannot help you more here, as I am not a programmer.

PE5PVB, 2023-09-16:
Ok. So the LIC is never displayed or sometimes? I'm not sure what happens on your side.

andimik, 2023-09-16:
See above, I did around 50 tests. **All** tests were successful with *redsea*, but only a few on TEF.

I only can say, the LIC field is not reliable as there is a bug.

PE5PVB, 2023-10-01:
Found it. Please test.

andimik, 2023-10-02:
It's much better, but there is still a bug.

When I am scanning in *Advanced RDS view*, a station here with LIC 0x00 will **never** show the LIC *unknown*. I need to **leave** the *Advanced RDS view* to *normal FM view* and then go again to *Advanced RDS view* and LIC *unknown* is displayed immediately.

PE5PVB, 2023-10-03:
Can you make a video of reproduction?

andimik, 2023-10-04:
[VID_20231004_222048.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/7089e3bc-30fd-4f3e-a6b6-a28ed9390dfd)

PE5PVB, 2023-10-05:
I see a lot of errors in the RDS stream. I checked it with our local transmitters, but couldn't reproduce it.

andimik, 2023-10-05:
Don't focus on the errors, the same result is when the signal is error-free (I can repeat the video of course). Please have a look at the LIC variable, as 0x00 is a value in this case (undefined).

PE5PVB, 2023-10-05:
Do they send 0x00 as value with LIC addressing?

andimik, 2023-10-05:
Yes, LIC with 0x00.

This video was taken with BER 0%.

[VID_20231005_205926.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/503c2606-0a7d-47d6-b3d4-5fa6fcee4f44)

```
10:26:49 5209 1460 3100 0000
10:26:49 5209 1460 3100 0000
10:26:50 5209 1460 3100 0000
10:26:50 5209 1460 3100 0000
```

[image]

```
$ cat ./5209_-_2023-06-24_10-26-38.spy | redsea -i hex -E | grep 1A
{"pi":"0x5209","group":"1A","bler":15,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":6,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
{"pi":"0x5209","group":"1A","bler":0,"has_linkage":false,"language":"Unknown","prog_type":"Information","tp":true}
```

see http://andimik.bplaced.net/sendinfo.php?id=250 as well.

andimik, 2023-10-05:
And I checked my own RDS archive (taken during vacations in several countries), which radio stations are using LIC 0x00.

```
$ egrep -rni -e "LIC  = 00" *.txt | grep LIC
2205_-_2020-08-21_17-37-36.txt:17:LIC  = 00
2311_-_2020-08-21_17-47-48.txt:17:LIC  = 00
2431_-_2020-08-21_17-35-56.txt:17:LIC  = 00
2D04_-_2020-08-21_18-25-36.txt:17:LIC  = 00
5209_-_2022-01-29_14-22-42.txt:17:LIC  = 00
5209_-_2023-06-24_10-28-28.txt:17:LIC  = 00
5218_-_2023-07-30_16-47-44.txt:17:LIC  = 00
5365_-_2023-09-14_16-38-18.txt:17:LIC  = 00
53C5_-_2023-06-18_14-11-10.txt:17:LIC  = 00
53C5_-_2023-09-14_16-34-14.txt:17:LIC  = 00
942B_-_2022-06-26_13-14-30.txt:17:LIC  = 00
942B_-_2022-06-26_13-16-20.txt:17:LIC  = 00
B201_-_2022-05-07_20-52-42.txt:17:LIC  = 00
B201_-_2023-06-30_18-59-46.txt:17:LIC  = 00
B202_-_2022-05-07_21-04-54.txt:17:LIC  = 00
B202_-_2023-06-30_19-06-14.txt:17:LIC  = 00
B203_-_2022-01-28_20-21-27.txt:17:LIC  = 00
B203_-_2022-05-07_20-51-07.txt:17:LIC  = 00
B203_-_2022-05-07_21-06-36.txt:17:LIC  = 00
B203_-_2023-06-30_19-04-03.txt:17:LIC  = 00
B317_-_2022-05-07_20-49-14.txt:17:LIC  = 00
C319_-_2023-06-18_14-47-12.txt:17:LIC  = 00
C319_-_2023-09-14_16-43-26.txt:17:LIC  = 00
```
So stations from 
- Hungary (Petőfi Rádió, Bartók Rádió, Kossuth Rádió), 
- Czech Republic (Radio Frekvence 1, Signal, Radio Beat, Radio Evropa 2), 
- Italy (RAI Isoradio, see logs above, Radio Gioconda, RTL102,5), 
- Croatia (HRT Radio Rijeka)

PE5PVB, 2023-10-06:
I think I found it. Please test

andimik, 2023-10-07:
No this https://github.com/PE5PVB/TEF6686_ESP32/commit/8c65f26785f90e624c256dd09b533dc8190edc23 did not fix it.

Still not able to display LIC *Unknown* when scanning in **Advanced RDS view**.

Seems you do not share the LIC variable in Advanced RDS view whereas in the normal view it works.

PE5PVB, 2023-10-07:
`Seems you do not share the LIC variable in Advanced RDS view whereas in the normal view it works.`

I'm not sure what you mean. Please make a video.

andimik, 2023-10-08:
The video is above.

Do all variables for LIC (has LIC, old LIC, etc) exist in normal and advanced view?

andimik, 2023-10-08:
During some tests (scanning in advanced RDS view) I realized that the following worked:

LIC *N/A* -> *Catalan* (due to bad reception) -> *unknown* (when signal was ok)

So the field is only updated when LIC has another value than 0x00.

andimik, 2023-10-09:
Yes, now it works :-)

Fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/cc8b43eb662100464c42ffc0003b8391ac685f06

closed 2023-10-09 by andimik