PE5PVB/TEF6686_ESP32 #266: Add RT-A/B-flag

lxsxl, 2023-09-07:
If I remember correctly there somewhen was a request on Discord to add RT-A/B-flag. This can be useful as sometimes the flag toggles while RT-content remains the same. Hopefully this can be added easily to main screen and/or advanced RDS-screen?

PE5PVB, 2023-09-10:
I can do that, but where to put it in advanced RDS view?

andimik, 2023-09-10:
I think two dots are enough

lxsxl, 2023-09-11:
I agree with Andis idea (two dots), a green dot for the currently displayed text, the other on in red.

Btw: Could you please again extend the RT-scrolling for about 5 characters on advanced RDS-screen? In case of very long texts it is still cut off a bit too early.

Thx!