PE5PVB/TEF6686_ESP32 #268: The PS line hurries the display of information due to which the dynamic PS is not displayed correctly

Yurij20, 2023-09-12:
Thanks for improving the operation of the RDS, but still, when the error mode is disabled, the PS line hurries up the display of information, as a result of which, if the reception of the RDS is uncertain, the PS is still displayed with errors.  You can make it so that when the error mode is disabled, information in the PS and RT lines is displayed only when it is fully loaded, both on car radios and in your basic firmware versions 1.16 and 2.07 from the v.2 tuner.  Thank you

PE5PVB, 2023-09-12:
What do you exactly mean? Wrong chars or missing chars?

PE5PVB, 2023-09-12:
The scrolling "end" is based on the initial textlength. When scrolling is busy and RT text gets updates, it will still end on the initial length. At a new scrolling start, the new length will be taken. I can't change this.

Yurij20, 2023-09-12:
> What do you exactly mean? Wrong chars or missing chars?

As a guide, in the RDS analyzer window, when the red dots are blinking, the PS line is updated unclearly and with missing characters.  It is better to make sure that it is updated when RDS reception occurs without errors and all 4 dots are confidently lit in green.  In this situation PS works fine.
[Video ](https://youtu.be/DlhwHVx1Pvg?si=stzTIkZ7onFYaouk)

PE5PVB, 2023-09-12:
Currently I check if the TEF says that the RDS blocks used for the PS are error free. If they are, then PS is processed. Maybe it will help you to turn on RDS buffer. I almost never see any PS errors when level is set to lowest value.

Yurij20, 2023-09-12:
> Currently I check if the TEF says that the RDS blocks used for the PS are error free. If they are, then PS is processed. Maybe it will help you to turn on RDS buffer. I almost never see any PS errors when level is set to lowest value.

In the menu I only see the RT buffer enabled.  It does not help.  There is no such item for the RDS buffer.

PE5PVB, 2023-09-12:
Sorry, it's called RDS filter.

Yurij20, 2023-09-12:
I turned on the RDS filter.  The problem still remains

PE5PVB, 2023-09-12:
Can you post a video with what the problem is?

Yurij20, 2023-09-12:
> Can you post a video with what the problem is?

https://youtu.be/DlhwHVx1Pvg?si=AfT3cK5Z8Cqh9QDm

PE5PVB, 2023-09-12:
Ah I now see what the problem is. PS is transmitted in 4x2 bytes. I currently only replace the correct received bytes and after 8 bytes refresh the screen. I'll change this behaviour.

Yurij20, 2023-09-12:
> Ah I now see what the problem is. PS is transmitted in 4x2 bytes. I currently only replace the correct received bytes and after 8 bytes refresh the screen. I'll change this behaviour.

Thank you!) I’m really looking forward to the update with the fix

PE5PVB, 2023-09-19:
When I have time.

PE5PVB, 2023-10-01:
I did add a clear of old PS after each rebuild of PS text. Please test.

Yurij20, 2023-10-01:
I checked, alas, the display has become worse (. Now I’ll take a video comparing the work of PS on the latest firmware 1.16 and 2.0

PE5PVB, 2023-10-01:
Problem is, I can't reproduce it with the stations I can receive (and lots of them are using dynamic PS. Please take this discussion to Discord and check if others have this problem as well. Possibly your radio station is just changing PS too fast.

Yurij20, 2023-10-01:
This happens on all stations with mediocre reception.  Here's a video comparison.  Now there is a suspicion that this may be due to the increased sensitivity of RDS coding in this firmware compared to 1.21, for example, because where 2.0 displays incorrectly, in 1.21 RDS reception stops and the RDS indicator blinks or goes out.  If possible, it's worth trying to lower it.  Please cancel the previous change to clear the PS line.  Below is a video comparing the operation of PS lines

https://youtu.be/2mn5qbQl4mc?si=XwuXmQ2q538wIPh-

PE5PVB, 2023-10-01:
I like to have a MPX recording from an SDR in .WAV at 192kHz to reproduce this. It also looks like the signalstrength is very low, which can cause errors in RDS data. Errorcorrection on RDS is not 100% flawless.

Yurij20, 2023-10-01:
Yes, you are absolutely right, PS does not work correctly with weak reception.  If it is strong, there are no problems.  It’s just that in firmware versions 1.21 1.15, with poor reception, there are practically no errors; PS data is either clearly displayed or not displayed at all.  In 2.0, alas, the situation is worse, even when the receiver is turned on or tuned to the reception frequency, information is typed onto the line in parts.  I made a second video with an exact display of the problem on another radio station by walking around the room with the receiver.  There is a proposal to compare the source codes responsible for the operation of PS lines in firmware 1.15 and 2.0.  If there are differences, move from 1.15 to 2.0.  Maybe this will fix the problem.  I just travel a lot with the receiver and it’s important that the PS displays the highest quality possible in difficult reception conditions, just like in firmware 1.15.  I'll make an MPX recording and send it later, it's not possible yet.  Below is another video with a more accurate display of the problem
https://youtu.be/dK2xeo4eSAA?si=z35qqVecmepNWylX

PE5PVB, 2023-10-01:
Versions before v2.00 have a total different PS decoder. With this method some older RDS encoders' PS were not detected the right way. This has been changed, but side-effect is the problem you're encountering on weak signals with high-speed PS rotation.

Yurij20, 2023-10-01:
Ok, thanks for understanding the essence of the problem :) If you don’t mind, it’s possible to compile firmware 2.0 with a PS decoder from firmware 1.15.  I agree to accept all the ensuing consequences that you wrote about above, because...  For me, the PS decoder from earlier firmware works fine and I don’t see any problems with it :) Thank you)

PasilTuner, 2023-10-02:
I completely agree with the user Yurij20. Indeed, this PS decoding method, unlike the old one from previous firmware, works very incorrectly with poor reception. I personally noticed this when traveling to another city with a radio and it is very inconvenient. I would really like you to return the old PS encoding method from firmwares up to 2.00, for example from 1.15, because... The highest quality PS encoding without errors is very important when the signal is weak.

PE5PVB, 2023-10-03:
I found just one other problem, this has to do with detection of RDS Block. RDS decoder now only will run when detection of RDS block has no errors at all.

Yurij20, 2023-10-03:
Okay, please compile the firmware on Discord) I’ll test it

Yurij20, 2023-10-08:
Hello, I tested today's firmware release, alas, nothing has changed regarding the operation of the PS line(. I made the same comparison with firmware 1.21. I kindly ask you to return to the old PS decoder from the old firmware

PE5PVB, 2023-10-08:
Roll-back is not an option. But, I did some more improvements. Please check.

Yurij20, 2023-10-08:
I checked, unfortunately the situation has not changed

PE5PVB, 2023-10-08:
In this case I need a MPX recording to investigate.

Yurij20, 2023-10-08:
Okay, I’ll do it soon, then I’ll unsubscribe and post it

Yurij20, 2023-10-10:
Hello, I made 2 recordings of two radio stations on which I showed the essence of the problem.  I hope I wrote it down correctly.
https://disk.yandex.ru/d/OOM1pT7iDBPnPQ

PE5PVB, 2023-10-10:
I checked both files with the current firmware but all looks good on my receiver.

https://youtu.be/clIm099mmXQ?si=vvWm1e26VST9tby6

https://youtu.be/GLyP_cxWjqA?si=DAXd15anJi_UscXW

Yurij20, 2023-10-10:
Yes, everything is correct here, I apologize, when I wrote it down, the reception was confident because...  I’m in another apartment, there’s a PS encoder here and it works great with this firmware.  The problem is only with poor reception, where the decoder of old firmware 1.21, for example, works well, but in 2.0 it works unclearly.  I made another recording of another station where at the moment the PS is rotating with an error, please check
https://disk.yandex.ru/d/1mgyfEX0nhnUZw

andimik, 2023-10-10:
I also have checked the two files, the BER is 0. PS is decoded normally in `redsea` and `RDSSpy`, see below.

[Bildschirmaufzeichnung vom 10.10.2023, 20:55:58.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/8e1175f5-839a-4c2b-8b80-287d192bc634)

```
20:58:04; 7244 06E8 E396 2043   0A  1 10111 01000  11100011 10010110  00100000 01000011  227 150  32  67  '..' ' C'
20:58:04; 7244 06E9 8C96 4F4D   0A  1 10111 01001  10001100 10010110  01001111 01001101  140 150  79  77  '..' 'OM'
20:58:04; 7244 06EA E396 4544   0A  1 10111 01010  11100011 10010110  01000101 01000100  227 150  69  68  '..' 'ED'
20:58:04; 7244 06EF 8C96 5920   0A  1 10111 01111  10001100 10010110  01011001 00100000  140 150  89  32  '..' 'Y '
20:58:04; 7244 06E8 E396 2043   0A  1 10111 01000  11100011 10010110  00100000 01000011  227 150  32  67  '..' ' C'
20:58:05; 7244 06E9 8C96 4F4D   0A  1 10111 01001  10001100 10010110  01001111 01001101  140 150  79  77  '..' 'OM'
20:58:05; 7244 06EA E396 4544   0A  1 10111 01010  11100011 10010110  01000101 01000100  227 150  69  68  '..' 'ED'
20:58:05; 7244 06EF 8C96 5920   0A  1 10111 01111  10001100 10010110  01011001 00100000  140 150  89  32  '..' 'Y '
20:58:05; 7244 06E8 E396 3130   0A  1 10111 01000  11100011 10010110  00110001 00110000  227 150  49  48  '..' '10'
20:58:05; 7244 06E9 8C96 322E   0A  1 10111 01001  10001100 10010110  00110010 00101110  140 150  50  46  '..' '2.'
20:58:06; 7244 06EA E396 3520   0A  1 10111 01010  11100011 10010110  00110101 00100000  227 150  53  32  '..' '5 '
20:58:06; 7244 06EF 8C96 464D   0A  1 10111 01111  10001100 10010110  01000110 01001101  140 150  70  77  '..' 'FM'
20:58:06; 7244 06E8 E396 3130   0A  1 10111 01000  11100011 10010110  00110001 00110000  227 150  49  48  '..' '10'
20:58:06; 7244 06E9 8C96 322E   0A  1 10111 01001  10001100 10010110  00110010 00101110  140 150  50  46  '..' '2.'
20:58:06; 7244 06EA E396 3520   0A  1 10111 01010  11100011 10010110  00110101 00100000  227 150  53  32  '..' '5 '
20:58:07; 7244 06EF 8C96 464D   0A  1 10111 01111  10001100 10010110  01000110 01001101  140 150  70  77  '..' 'FM'
20:58:07; 7244 06E8 E396 3130   0A  1 10111 01000  11100011 10010110  00110001 00110000  227 150  49  48  '..' '10'
20:58:07; 7244 06E9 8C96 322E   0A  1 10111 01001  10001100 10010110  00110010 00101110  140 150  50  46  '..' '2.'
20:58:07; 7244 06EA E396 3520   0A  1 10111 01010  11100011 10010110  00110101 00100000  227 150  53  32  '..' '5 '
20:58:07; 7244 06EF 8C96 464D   0A  1 10111 01111  10001100 10010110  01000110 01001101  140 150  70  77  '..' 'FM'
20:58:07; 7244 06E8 E396 3130   0A  1 10111 01000  11100011 10010110  00110001 00110000  227 150  49  48  '..' '10'
20:58:08; 7244 06E9 8C96 322E   0A  1 10111 01001  10001100 10010110  00110010 00101110  140 150  50  46  '..' '2.'
20:58:08; 7244 06EA E396 3520   0A  1 10111 01010  11100011 10010110  00110101 00100000  227 150  53  32  '..' '5 '
20:58:08; 7244 06EF 8C96 464D   0A  1 10111 01111  10001100 10010110  01000110 01001101  140 150  70  77  '..' 'FM'
20:58:08; 7244 06E8 E396 3130   0A  1 10111 01000  11100011 10010110  00110001 00110000  227 150  49  48  '..' '10'
```

```
$ redsea -E -r 192k -f 1025.wav | grep ps | jq -r ".ps"
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
102.5 FM
102.5 FM
102.5 FM
102.5 FM
102.5 FM
102.5 FM
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
102.5 FM
102.5 FM
102.5 FM
102.5 FM
102.5 FM
102.5 FM
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
102.5 FM
102.5 FM
102.5 FM
102.5 FM
102.5 FM
102.5 FM
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
 COMEDY 
```

```
$ redsea -E -r 192k -f 1030.wav | grep ps | jq -r ".ps"
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
RADIO   
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
SHANSON 
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
103.0 FM
RADIO   
RADIO   
```

PE5PVB, 2023-10-10:
Thanks for the last file, I modified the software. Please check.

Yurij20, 2023-10-11:
Thank you) I'm waiting for a new release of firmware with corrections.  I'll check then

Yurij20, 2023-10-11:
I checked the update now, it really got better after the introduction of the PS buffer.  But the problem remains that when PS starts, at the moment of turning on the receiver or tuning into a radio station, PS still types the first lines in parts or with errors.  This happened before, even with static PS, I forgot to mention this earlier.  Please pay attention to this.  As an option, I propose to make the PS buffer work directly when an RDS block is detected, i.e. at the moment the receiver is turned on, tuned to a station, etc.  Apparently, old firmwares work in this mode, for example 1.21

Yurij20, 2023-10-12:
I checked the update now, it really got better after the introduction of the PS buffer.  But it still misses quite a lot of errors with weak reception, where in the old firmware 1.21 ps the encoder does not rotate.  There is still a problem that when starting PS, at the moment of turning on the receiver or tuning into a radio station, PS still types the first lines in parts or with errors.  This happened before, even with static PS, I forgot to mention this earlier.  Please pay attention to this.  As an option, I propose to make the PS buffer work directly when an RDS block is detected, i.e. at the moment the receiver is turned on, tuned to a station, etc.  Apparently, old firmwares work in this mode, for example 1.21

PE5PVB, 2023-10-12:
Good idea. I've added a menu option where you can disable this Fast PS. PS is then only shown when 2 cycles of the same PS are received.

Yurij20, 2023-10-12:
Tested, ps now starts perfectly without errors in any reception conditions).  But when rotating the dynamic ps, it still misses a lot of errors, as I said before with weak reception and interference.  There is another proposal in the disabled Fast ps mode when rotating the dynamic PS to allow each rotation of the PS only when then when 2 cycles of the next part of the dynamic PS are received.  Is it possible?  Then the possibility of errors will be extremely small

PE5PVB, 2023-10-12:
That is already the case. It first gets the PS one time, then a second time. When both are equal then it's displayed. I'm out of options. Only difference I can guess is that the older version numbers are compiled with another display clockspeed, so interference on the FM band is slightly moved to another frequency.

Yurij20, 2023-10-12:
Good) Apparently my reception conditions have worsened in my apartment, I’ll test it more accurately tomorrow and report back.  And it’s still better than it was before.  Regarding interference from the display, your compilation gives interference on 87.3 and 101.8.  I have a Chinese device.  Early versions did not produce interference in the fm range.  Can you tell me at what frequency the display should be compiled for my device so that there is no interference on FM from the display?

PE5PVB, 2023-10-12:
With the older version you should have interference at around 90.0MHz. I will close this issue because there is nothing more to do for me.

closed 2023-10-12 by PE5PVB

Yurij20, 2023-10-18:
I apologize for bringing up this topic again.  I compared the work with old firmware and came to the conclusion that in PS the line rotates even with weakened RDS reception and often collects errors.  In older firmware, under the same reception conditions, with a weakened RDS signal, the PS line does not rotate at all and resumes rotation when strong RDS reception is achieved.  There is a proposal, if possible, to set a ban on dynamic PS rotation when RDS reception is weakened, thereby making the PS line rotate only with strong 100% RDS reception.  Thank you