PE5PVB/TEF6686_ESP32 #275: Screensaver: durations >60s

AaronJowitt, 2023-09-20:
A few long options would be useful. 30 minutes, 1 hour.

andimik, 2023-09-20:
You can change the sources if you want. Therefore, this is not an issue.

https://github.com/PE5PVB/TEF6686_ESP32/blob/e605fc2e5a732add5925fc4cb72bf321d27bb92e/TEF6686_ESP32.ino#L159

mx4zxc, 2023-09-20:
@AaronJowitt 
I agree that we should have more choices, so we could really use it as a sleep timer. But as andimik says, this works as intended.

@andimik 
I've played around with that part a bit, but I could never get the value to be higher than 200 sec. So I think there are some limitations there.
If you know of a solution to choose 30 minutes or more, I'm interested in knowing how.

andimik, 2023-09-20:
But was is the intention to have such a long time?

mx4zxc, 2023-09-20:
I use the radio on the bedside table when I go to bed. Many radios have this function where they switch off automatically after a specified time.
This radio also has this option called "Deep sleep", but becomes a bit pointless in this setting when you can only choose a maximum of 60 seconds.
It would be nice to have this option on this radio too if it is possible. But the radio uses little power anyway, so it's not that important :)

andimik, 2023-09-20:
Have you tried 255 sec?

mx4zxc, 2023-09-20:
[image]
It seems like the max is 299 for some reason.
Could there be a way to increase that value?

andimik, 2023-09-20:
Yes, you need to replace `byte`. But I am not a programmer...

mx4zxc, 2023-09-20:
I'm certainly not a programmer either, but I like to explore a bit.
I was able to change the value from seconds to minutes. Haven't tested much, but a minute seemed fine, testing 10 min now.

File: constants.h 
line: 15
Changed from:
#define TIMER_SCREENSAVER_BASE     1000000
to:
#define TIMER_SCREENSAVER_BASE     60000000

andimik, 2023-09-21:
But again, what is the benefit of such a long time? The screen goes off, but sound is still available.

mx4zxc, 2023-09-22:
Yes, it is very useful the way it works in the first place.
I tested a bit more yesterday and it looked like everything worked fine up to 30 minutes, but when I tried 50, 60 and 70 minutes, the radio became unstable and it went into a restart loop. So it's probably not the right way to do it.

mx4zxc, 2023-09-22:
@AaronJowitt
The only simple solution I can suggest is to change the value from seconds to minutes.
You need to change some values in at least two files to make this happen.

First you need to change this line in constants.h:
from:
#define TIMER_SCREENSAVER_BASE 1000000
to:
#define TIMER_SCREENSAVER_BASE 60000000UL

Then you have to change this in your language in language.h:
from:
"sec.", // 92
to:
"min.", // 92

I don't know how correct this solution is, but testing so far works fine so far on my radio.
Just keep in mind that the shortest value will then be 1 minute and the longest I have got to work is 60 minutes.