PE5PVB/TEF6686_ESP32 #286: Wrong battery symbol color resp. no values

andimik, 2023-10-02:
The battery symbol color depends if WiFi was on during start or not.

# issue 1

When WiFi is not connected during start, but activated afterwards, it has a red border color, which might be misunderstanding (low current). So, I propose to have this symbol in grey, too.

# issue 2

There is no value after WiFi is switched off, the battery border color is grey. Workaround: reboot.

## Remark:

It has a (correct) grey border color, when WiFi is connected during start.

andimik, 2023-10-06:
[image]

PE5PVB, 2023-10-06:
I've changed something in this matter, but have no possibility to test it. Please check

andimik, 2023-10-07:
No, there was no improvement, still red border. Tested with latest 8c65f26 dated yesterday. 

This was taken right after configuring WiFi and connecting to the WiFi.
[image]
After a restart with Wifi on it became grey.

When I switch off Wifi the border is still grey with no value inside.

[image]

I need to restart in order to get the value back.

Seems you cannot change the variable during a session?

PE5PVB, 2023-10-07:
Possibly both are fixed after the WiFi issue in the other issue. Please check.

andimik, 2023-10-08:
No, not fixed, still showing red border.

PE5PVB, 2023-10-08:
Please check again. If problem is still there, please make a new short movie for reproduction.

andimik, 2023-10-08:
Yes, fixed.

closed 2023-10-08 by andimik