PE5PVB/TEF6686_ESP32 #287: Scan versus SW Band

andimik, 2023-10-02:
The scan function in SW is only working if *AUTO* was selected on **another** band and then switched to SW again. Pressing the *mode* button will change from *MAN* to *BAND*

PE5PVB, 2023-10-03:
I can't reproduce that problem. Can you make a movie of reproduction?

Mike86-CZE, 2023-10-04:
It's more of a misunderstanding of the controls than a bug.  In the SW range, if you want to use auto scan, hold down the tuning knob for a long time.

PE5PVB, 2023-10-04:
So what is your suggestion? Create a menu-item where this can be set?

Mike86-CZE, 2023-10-04:
Rather just explain the controls to the user (wiki?).  I think many people don't know how to activate AUTO/BAND in SW range.

andimik, 2023-10-04:
Oh, interesting, yes, it's a good idea.

closed 2023-10-10 by andimik