PE5PVB/TEF6686_ESP32 #288: TMC icon not reliable

andimik, 2023-10-02:
The TMC icon is not reliable as it indicates too few times TMC.

For example **redsea** gives the following output:

```
$ nc 127.0.0.1 7376 | redsea -hE  | grep tmc
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Resurfacing work.","direction":"both","event_codes":[704],"extent":"-0","location":31401,"update_class":11,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"gap":3,"service_id":0}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Queuing traffic. Single alternate line traffic.","direction":"both","event_codes":[108,513],"extent":"-1","location":51040,"update_class":1,"urgency":"U"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"gap":3,"service_id":0}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Closed. Construction work.","direction":"both","event_codes":[401,803],"extent":"-0","location":42196,"update_class":5,"urgency":"U"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Lane closed. Construction work.","direction":"single","event_codes":[500,803],"extent":"-0","location":44365,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"gap":3,"service_id":0}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Closed. Construction work.","direction":"both","event_codes":[401,803],"extent":"-0","location":44401,"update_class":5,"urgency":"U"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Message cancelled.","direction":"single","event_codes":[625],"extent":"-2","location":31887,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Message cancelled.","direction":"single","event_codes":[625],"extent":"-2","location":31887,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Message cancelled.","direction":"single","event_codes":[625],"extent":"-2","location":31887,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"gap":3,"service_id":0}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Roadworks. Maintenance work.","direction":"both","event_codes":[701,703],"extent":"-0","location":32572,"update_class":11,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"gap":3,"service_id":0}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Roadworks.","direction":"both","event_codes":[701],"extent":"-0","location":30348,"update_class":11,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Roadworks. Single alternate line traffic.","direction":"both","event_codes":[742],"extent":"-0","location":32052,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Roadworks. Single alternate line traffic.","direction":"both","event_codes":[742],"extent":"-0","location":32052,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Roadworks. Single alternate line traffic.","direction":"both","event_codes":[742],"extent":"-0","location":32052,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Contraflow.","direction":"both","event_codes":[517],"extent":"-3","location":44419,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Contraflow.","direction":"both","event_codes":[517],"extent":"-3","location":44419,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Contraflow.","direction":"both","event_codes":[517],"extent":"-3","location":44419,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"gap":3,"service_id":0}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Single alternate line traffic. Construction work.","direction":"single","event_codes":[513,803],"extent":"-0","location":38953,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Closed. Maintenance work.","direction":"single","event_codes":[401,703],"extent":"-0","location":35929,"update_class":5,"urgency":"U"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"gap":3,"service_id":0}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Roadworks.","direction":"both","event_codes":[701],"extent":"-0","location":38951,"update_class":11,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Roadworks. Single alternate line traffic.","direction":"both","event_codes":[742],"extent":"-0","location":6800,"update_class":5,"urgency":"none"}},"tp":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Roadworks. Single alternate line traffic.","direction":"both","event_codes":[742],"extent":"-0","location":6800,"update_class":5,"urgency":"none"}},"tp":false}
[... cut ...] p":false}
{"pi":"0xA213","group":"3A","bler":0,"open_data_app":{"app_name":"RDS-TMC: ALERT-C","oda_group":"8A"},"prog_type":"No PTY","tmc":{"system_info":{"is_encrypted":false,"is_on_alt_freqs":true,"location_table":1,"scope":["national"]}},"tp":false}
{"pi":"0xA213","group":"8A","bler":0,"prog_type":"No PTY","tmc":{"message":{"description":"Roadworks. Maintenance work.","direction":"both","event_codes":[701,703],"extent":"-0","location":42435,"update_class":11,"urgency":"none"}},"tp":false}
```

PE5PVB, 2023-10-03:
I'm not sure what I see in your outputtext. I only check for a specific bit in Block B in RDS Group 8A. So, it can take a few seconds before TMC flag is shown.

andimik, 2023-10-06:
8A?

You need to check b12, b13, b14 in  1A.

andimik, 2023-10-06:
[image]

PE5PVB, 2023-10-06:
I think I found to fault. I also added detection on 1A. But this is not used on all stations. Please test.

andimik, 2023-10-06:
Yes, all stations here which have TMC have now the icon within one second :+1: 

Therefore, I can close it.

closed 2023-10-06 by andimik