PE5PVB/TEF6686_ESP32 #290: Some fields in AF/EON view not cleared on frequency change

andimik, 2023-10-02:
In AF/EON view some fields (PI and current received frequency) are not cleared at frequency change (via *xdr-gtk*), and texts have layers of all used pixels.

Expectation: all fields and lines cleared.

See video.

[VID_20231002_230428.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/0313f90b-03af-4bc3-aeb3-5b475089e75c)

PE5PVB, 2023-10-03:
You should not be able to tune in the AF/EON screen. I changed the software so it will bring you back to the Advanced RDS view first. Please test.

andimik, 2023-10-05:
> I changed the software so it will bring you back to the Advanced RDS view first.

No, not with the up/down icons (see screenshot). You also have to implement it ...

[image]

[image]

PE5PVB, 2023-10-06:
You are correct. I forgot the seek buttons. I have changed this. Please test

andimik, 2023-10-06:
Yes, now it works, fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/4b689f23b14ff8cb7b8e611d86c7cc92ac894f84. Therefore, issue closed.

closed 2023-10-06 by andimik