PE5PVB/TEF6686_ESP32 #291: Wifi connection not closed

andimik, 2023-10-06:
I realized that switching off Wifi in the menu does not work in the past commits. It's still connected although it shouldn't.

The screenshot shows 

> Wi-Fi active: Off

but the Stationlist client IP is still shown, which makes no sense. And I can still connect in *xdr-gtk*.

[image]

After a reboot Wifi is really off.

PE5PVB, 2023-10-06:
You are right. I've changed this.

andimik, 2023-10-07:
**Not** confirmed, **not** fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/e0ac42567568af41faf7f11292221c3a54bf8399

TEF does **not** close the WiFi connection, Stationlist client got a `-` then and the Wifi symbol disappeared (battery symbol still grey). **But you still can connect with `xdr-gtk`.**

PE5PVB, 2023-10-07:
Should be fixed now. Please test.

andimik, 2023-10-08:
Yes, fixed in https://github.com/PE5PVB/TEF6686_ESP32/commit/9227fe77363e1c24e00129f00912bba64dcb6ed6

Closing the issue.

andimik, 2023-10-08:
:warning: 
Note: Now the behaviour is different than before. Wifi is **not automatically connected** after configuring. The message `CONNECTED` does **not** appear anymore and in the menu Wi-Fi active is off.

I need to set WiFi active to On in order to connect.

PE5PVB, 2023-10-08:
You are right. Should be fixed now.

closed 2023-10-08 by andimik