PE5PVB/TEF6686_ESP32 #292: ECC field will not be updated

andimik, 2023-10-07:
During bad reception a station was identified as ECC unknown by mistake, instead of Italy.

[image]

But the main problem is, that the field will not be updated until re-tuning to the same frequency. So it's static.

[image]

Please note, that it will be difficult to reproduce it.

andimik, 2023-10-07:
I now had that again with ECC *unknown* which was never updated to the real value (this time Croatia) before I left the Advanced RDS view. So it can be reproduced.

PE5PVB, 2023-10-07:
Hmm.. I checked the code but it should be updated every cycle. I can't find the cause of this and haven't been able to reproduce this. Only thing I found is that during scanning it was not updated.

andimik, 2023-10-08:
That's what I am talking about all the time. If you scan in advanced RDS view, if will not update.

PE5PVB, 2023-10-08:
I've tested it with an RDS encoder. When in Advanced RDS view, and I change LIC or ECC in the encoder, it automaticly updates on the display, I really don't understand your problem.

andimik, 2023-10-08:
Which LIC number are you using? Please use 0x00, because for all other values it works.

andimik, 2023-10-08:
Which ECC number are you using? Please test it again with 0x00 (unknown).

PE5PVB, 2023-10-08:
ECC is a combination with the first digit of PI code, so I don't understand your question

PE5PVB, 2023-10-08:
Hmmm... I think I found the cause if this issue. Please test.

andimik, 2023-10-09:
After a start (as long as hasECC is false) the field is empty (not N/A).

[image]

As soon as one station with ECC was received, 

[image]

all following stations have the correct value.

[image]
__

closed 2023-10-09 by andimik