PE5PVB/TEF6686_ESP32 #293: CT field shows first UTC time and only then local time

andimik, 2023-10-07:
TEF will first show 23:13 (UTC) and then 0:13 (local) **when received**, see video.

It's **not** related to that station only, I also have the two times on public broadcasters (where the time is the DCF time). It first shows UTC and changes to local time one second later, see second video.

[VID_20231007_121409.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/8ae4656d-f0a0-4b5a-a65a-c9af58c7edf0)

[VID_20231007_123257.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/cff545d5-f616-491e-b073-56d6fb5ae8f6)

[image]

```
$ nc 127.0.0.1 7376 | redsea -hEp  | grep time
{"pi":"0x5293","group":"4A","bler":0,"clock_time":"2019-07-27T00:13:00+01:00","prog_type":"Rock music","tp":false}
{"pi":"0x5293","group":"4A","bler":0,"clock_time":"2019-07-27T00:13:00+01:00","prog_type":"Rock music","tp":false}
{"pi":"0x5293","group":"4A","bler":0,"clock_time":"2019-07-27T00:13:00+01:00","prog_type":"Rock music","tp":false}
{"pi":"0x5293","group":"4A","bler":0,"clock_time":"2019-07-27T00:13:00+01:00","prog_type":"Rock music","tp":false}
{"pi":"0x5293","group":"4A","bler":0,"clock_time":"2019-07-27T00:13:00+01:00","prog_type":"Rock music","tp":false}
{"pi":"0x5293","group":"4A","bler":0,"clock_time":"2019-07-27T00:13:00+01:00","prog_type":"Rock music","tp":false}
{"pi":"0x5293","group":"4A","bler":0,"clock_time":"2019-07-27T00:13:00+01:00","prog_type":"Rock music","tp":false}
{"pi":"0x5293","group":"4A","bler":0,"clock_time":"2019-07-27T00:13:00+01:00","prog_type":"Rock music","tp":false}
```

PE5PVB, 2023-10-07:
This is possibly caused by a double CT update. I reduced the CT updates to every minute. Does this help?

andimik, 2023-10-08:
General question: Why do you need to display both times?
If UTC is worth to display you should label it as `UTC`.

Every minute there is either UTC or (which is a new bug) a fake time.

See video. The station only transmits `2019-07-27T00:13:00+01:00` (no other time, no update), but each minute the first value is `00:14` instead of `23:13` like before in the first posting.

[VID_20231008_084534.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/a1534106-e3b3-4bd6-a9a2-d36503f1c444)

PE5PVB, 2023-10-08:
Yes, looks like a bug in the Arduino/compiler platform. I've changed it now. Only after receiving CT, time will be set.

andimik, 2023-10-08:
Now it will show `0:14`, `0:15` and so on, although the RDS CT time has not changed. 

I know this is unusual behavior, but I suggest to print the actual value and not calculated time ...

But did it work until last week (or so)?

PE5PVB, 2023-10-08:
Yes, it's regression on another bugfix. You can't have both. I now implemented your advice. Time is now shown as transmitted over RDS, not the internal RTC.

andimik, 2023-10-08:
For this particular station (Radio Freccia) this now won't work like before (even after several minutes you cannot see CT, although it's transmitted each 2 sec.), but I can live with it as nobody else than me will probably have such a situation.

On other radio stations after 1 minute the time is displayed.

[VID_20231008_202738.webm](https://github.com/PE5PVB/TEF6686_ESP32/assets/24510556/f87043e6-995b-4bac-a6a3-c05f26fa0113)

As no UTC is displayed now (only local time), I will close it.

closed 2023-10-08 by andimik