PE5PVB/TEF6686_ESP32 #296: TP and TA flag

mtv-alex, 2023-10-08:
Hello.
If the radio station transmits the TP flag, the receiver will display the TP flag. At the time of transmission of traffic information (transmission of TP/TA flags), the display will also display TP/TA flags.
But the RDS standard allows the transmission of only the TA flag (TP is turned off). By transmitting the TA flag, we inform the receiver that there is a switch to traffic information via EON. If the station transmits the TA flag, then the receiver firmware does not display this (TA flag).
[image]
[image]

PE5PVB, 2023-10-08:
I've changed the TA flag behaviour.

mtv-alex, 2023-10-09:
After installing the new firmware it worked. Thank you.

closed 2023-10-09 by mtv-alex

andimik, 2023-10-09:
Yes, I also tried it and it worked.

andimik, 2023-10-09:
Well, it shows `TA` for EON only stations, this could be misunderstanding, although it's correct. But we have the `EON` for this reason.