PE5PVB/TEF6686_ESP32 #298: AF method B regional variant detection not correct

andimik, 2023-10-11:
I fear that in some cases the R in front of the frequency is not correct. It should be visible in Method B, when the frequency transmits a local variant of the program. But I have it also on (nationwide) programs which do not have local variants at all (in Austria it's Ö1, see screenshot).

[image]

As far as I can remember it's quite easy to detect regional variants, as the sorting of the frequency pair is different then.

Maybe you should check looking at `buffer0` and `buffer1` why `af[af_counter].regional` becomes `true` in this case. I don't believe that in our case.

```
$ nc 127.0.0.1 7376 | redsea -hE | grep freq | jq -r ".alt_frequencies_b" > /tmp/oe1_af.txt
```

If there would be a regional variant the *redsea* log would include the term `regional_variants` (have checked on a program which has that).

# redsea log

[oe1_af.txt](https://github.com/PE5PVB/TEF6686_ESP32/files/12865614/oe1_af.txt)

# RDSSpy log

[A201_-_2023-10-11_07-49-03.txt](https://github.com/PE5PVB/TEF6686_ESP32/files/12865615/A201_-_2023-10-11_07-49-03.txt)

andimik, 2023-10-11:
Oh, seems I was right and `buffer0` and `buffer1` were not correctly analyzed.

Will test it in the evening.

lxsxl, 2023-10-11:
I'm sorry that I was not available due to holiday (what was fine though :-)). But now I flashed the latest beta version. And I would say that whenever I see R-indicators it is on stations with AF-method B. But for most of the stations that use RV the indicators "R" and also "non-R" are wrong.

Probably we can approach with one of the MPX-file that I sent you a few weeks ago (D6A4 / SWR4_KO_ from the received frequency 97,4):

On this frequency there are luckily only two AF-lists transmitted, so it's much more easier. One AF-list is for the in fact received frequency (97,4) and one for the neighbour-frequency (93,3). The AF-lists/AFs are transmitted as follows:

count: 9 / 97,4
97,4 / 93,6 (-> because of descending order within this pair it is RV)
97,4 / 106,3 (-> because of the ascending order within this pair it is non-RV)
97,4 / 107,4 (-> because of the ascending order within this pair it is non-RV)
93,3 / 97,4 (-> because of the ascending order within this pair it is non-RV)

count: 7 / 93,3
93,6 / 93,3 (RV)
93,3 / 97,4 (non-RV)
93,3 / 107,4 (non-RV)

The TEF actually displays 106,3 and 107,4 as RV (which are definitely non-RV), and this is reproducible. So it actually works but in some way wrong.

On those stations with more AF-lists than in this example all AF will be shown as non-RV. So it seems that there is also a problem when the number of AF-lists exceeds a critical threshold.

For the next step it would help to get this example working. Hopefully it will also help to be successful with the more complex cases...

lxsxl, 2023-10-11:
It seems that the switch from method A to method B works: I tuned to the frequency mentioned above and switched immediately to EON/AF-screen: The AFs appeared step by step but without "R". Step back to advanced RDS-screen and again to EON/AF-screen, then the "R"-indicators were present.

PE5PVB, 2023-10-11:
Hmm. there is some misunderstanding here. I thought in both cases it was region but only a different area. I will change it. to only when AF1 > AF2

PE5PVB, 2023-10-11:
Done. Please test.

lxsxl, 2023-10-11:
I will do...

lxsxl, 2023-10-11:
Oh, I'm sorry, I am not (or at least not yet) familiar with compiling. Can you add a compiled version? But anyway I will practise in compiling :-)

lxsxl, 2023-10-11:
just found it on discord, thx!

lxsxl, 2023-10-11:
Now there is no R-indicator at all any more.

lxsxl, 2023-10-11:
> I thought in both cases it was region but only a different area. I will change it. to only when AF1 > AF2

Can you please define "area" and "region"?

In RDS specification for AF-method B it is like this:
One single frequency transmits several AF-lists so that the receiver knows the matching AFs for the actually received frequency **and also** for other frequency of the same network. The RV-information is only meaningful within one single AF-list. So in two different AF-lists the information on RV may be different (because the head-frequencies of the two lists in this case belong to  different RVs). So it is essential to be sure, that you received the **complete** AF-list and did not skip some AF-pairs due to bad reception.

PE5PVB, 2023-10-11:
Currently I check for double current frequencies to detect AF method B.

I also check if AF1 = current frequency, and if this is the case, if AF1 > AF2 then R is attached to AF2.

lxsxl, 2023-10-11:
Okay, sounds good so far, but what about the example:

count: 9 / 97,4
97,4 / 93,6 (-> because of descending order within this pair it is RV)
97,4 / 106,3 (-> because of the ascending order within this pair it is non-RV)
97,4 / 107,4 (-> because of the ascending order within this pair it is non-RV)
93,3 / 97,4 (-> because of the ascending order within this pair it is non-RV)

count: 7 / 93,3
93,6 / 93,3 (RV)
93,3 / 97,4 (non-RV)
93,3 / 107,4 (non-RV)

Should the program discover the 93,6 als RV? Particularly as it is RV in both AF-lists? (Actually it doesn't...)

PE5PVB, 2023-10-11:
It can take some time to discover. Did you wait long enough?

lxsxl, 2023-10-11:
I think yes. The AFs were read in a few seconds, I waited one or two further minutes.

PE5PVB, 2023-10-11:
OK. Started from scratch. Now only frequencies marked RV on current frequency in RDS Spy are also marked with a R. Please check.

PE5PVB, 2023-10-11:
Only thing left to do is refresh AF screen as soon as Method B is detected. Another thing for tomorrow.

andimik, 2023-10-12:
AF refresh see https://github.com/PE5PVB/TEF6686_ESP32/commit/e16ffedbdcf9623e306399c804408e1bdff781ea

But this commot not tested so far. I guess it only has to do when reception is poor and method B is detected only after some seconds in AF/EON view.

But I only can receive one station with local variant, and this is displayed with R -> so I'll close that issue.