PE5PVB/TEF6686_ESP32 #300: RT details

lxsxl, 2023-10-11:
I would like to suggest a few changes on RT behaviour:

In case of flag toggle it is difficult to read the complete RT content (particularly if underscores are activated?). Actually you only see that the previous RT is replaced by <space> or underscores which scroll to the end of the full 64 characters. If it is fast enough sometimes at least a few last characters of the new RT are visible; the full new RT will not be visible until the 64 characters have passed completely. And it also may happen that again the next RT starts before you were able to read a full RT-content. So there are some ideas:
1. If RT-flag toggles, immediately restart RT from first character. OR
2. Respond to x0D-signal. I know that you said that it is questionable because some stations transmit further characters after x0D. But maybe you could check for x0D AND no other characters than <space>?

This is all only in case of deactivated RT-buffer (where scrolling to the end should be fine even on flag toggle).

And a second matter: In case of flag toggle, the new text immediately starts scrolling, so that it is hard to read the first word(s). The RT starts scrolling even if it is short enough so that scrolling would be unnecessary. It would be better if scrolling not starts until a critical number of characters has been received.

PE5PVB, 2023-10-11:
OK. What I changed: When radiotext changes, ticker is restarted.
I'll look into your second question tomorrow.
Scrolling already has a leading holdtime.

PE5PVB, 2023-10-12:
I've added your 2nd request. When station uses 0x0d, extra underscores are removed.

lxsxl, 2023-10-12:
Looks very fine now, thanks!

closed 2023-10-12 by PE5PVB