PE5PVB/TEF6686_ESP32 #302: Unreadable character ' (apostrophe)

mtv-alex, 2023-10-12:
Hello.
I noticed that if the name contains the symbol ( ' - apostrophe), the receiver cannot display it.
[image]
[image]

PE5PVB, 2023-10-12:
Can you get me a MPX recording of that station?

mtv-alex, 2023-10-12:
https://disk.yandex.ru/d/XWXaOIn7XbzXAg
This is an edited file from which you can extract the MPX (via SDR Console). F=87.90 MHz.

PE5PVB, 2023-10-12:
There is no MPX in that file. But I think I found it. Seems that ` was not in use in the original RDS tables. Please check.

PE5PVB, 2023-10-12:
I checked again the latest RDS table from the EBU, ` is not assigned. It's very weird this broadcaster uses it.

mtv-alex, 2023-10-12:
Yes, this is a very interesting symbol. The sign is number 96. According to the table, this means... two vertical lines. Those. According to logic, everything showed correctly.
I can assume that a different version of the symbol table is being used. We have several of them: default, G0, G1 and G2.
[image]
[image]

mtv-alex, 2023-10-12:
But in this table it is present under the same number: 0х60
[image]

PE5PVB, 2023-10-13:
Can you get me a full G0 RDS characterset (not as a picture)?

mtv-alex, 2023-10-13:
I only found such old tables: https://pira.cz/rds/rdschars.pdf
I don't have any new ones. There is such a symbol table. But it is for 2009 and changes could have been made during this time.
As I understand it, an ASCII character table is used for the display. Looking at the updated table, I see that the symbol ` is hidden under the number 96 (DEC). https://www.ascii-code.com/
And apparently changes were made to this table, which were later transferred to the RDS standard.
For a detailed analysis, it is better to use RDS Spy. You can see everything in it, which characters are passed to DEC, HEX.

PE5PVB, 2023-10-13:
Well. Currently I already have remapped 0x60 to `
If anyone can provide the complete ASCII lists for G0, G1 and G2 and give a explanation how to detect G0, G1 or G2 please let me know.

Please test the character and report back.

lxsxl, 2023-10-13:
The character-sets should be completely in the RDS specification (pdf-pages 118 - 138).

PE5PVB, 2023-10-13:
Can you please post the document you refer to? There are several different versions. I currently use ISO62106. I can't find any reference to the `-char.

lxsxl, 2023-10-13:
That's the document I meant. 0x60 is 0x2016 DOUBLE VERTICAL LINE (and not apostrophe, see pdf-page 121). I didn't find this kind of apostrophe in the tables, even not in the cyrillic part. But I didnt't read this whole thread, so I would agree with you that the broadcaster is using an old table (and DEVA is still using it too...). Standard-non-conformity is always a problem ...

PE5PVB, 2023-10-13:
Well, since no-one will use the double vertical bar, I will replace it with a `. Still waiting for confirmation of @mtv-alex if this solved his issue.

mtv-alex, 2023-10-14:
Yes, the problem is solved.
[image]

closed 2023-10-14 by mtv-alex