PE5PVB/TEF6686_ESP32 #303: Language code table

mtv-alex, 2023-10-12:
Hello.
Is the entire language code table used? In the RDS 1A group we can transmit the name of the country and the broadcast language. The radio station's broadcast language is set to 79 (HEX) = Belorussian. But the receiver does not recognize him - writes "unknown"
[image]
[image]

PE5PVB, 2023-10-12:
You can check the current LIC table in the bottom of the language.h file.
Please add the missing languages to it, or post the list here.

mtv-alex, 2023-10-12:
Yes, if you can, please add Belarusian language for the item LIC. According to the RDS table, it has the number: HEX = 79, DEC = 121.
Other major languages are present. Thank you

PE5PVB, 2023-10-12:
Do you have a full list of LIC languages with their address?

PE5PVB, 2023-10-12:
I already found a table and applied it. Please test.

mtv-alex, 2023-10-12:
Yes, I see that the language has been added in the "language.h" file. Thank's

mtv-alex, 2023-10-14:
Somewhere there are not enough languages, or a space. Or vice versa - overkill.
Where there is a Belarusian language, Armenian is shown; where Russian is Persian. The difference is 4 lines.

andimik, 2023-10-14:
Maybe it helps to add the numbers as comments again to the languages?

mtv-alex, 2023-10-14:
A comparative analysis of language codes according to the RDS standard was carried out with those codes that were introduced (language.h). All this is summarized in a table. Everything that is green matches the RDS standard, everything that is red does not. As I said, there are a few blank lines missing.
[LIC.xlsx](https://github.com/PE5PVB/TEF6686_ESP32/files/12906727/LIC.xlsx)

PE5PVB, 2023-10-14:
Thanks @mtv-alex . I just modified the table.

andimik, 2023-10-14:
And thanks for following my suggestion :-)

PE5PVB, 2023-10-14:
Thank ChatGPT for that

mtv-alex, 2023-10-15:
Everything is fixed Thank you!
[image]

closed 2023-10-15 by mtv-alex