PE5PVB/TEF6686_ESP32 #304: AF / regional versions

lxsxl, 2023-10-12:
I just tested the latest AF/RV programming. I would say: We are close to perfection, at least all RV that are in the AF-list for the tuned frequency are now marked as RV. But there are two things that should be adjusted:

1. It may be that one AF is in the list as RV AND ALSO as non-RV (e. g. 100,4 / 91,8 AND 91,8 / 100,4). This may happen when the same frequency (in this case 91,8) is used in two different sites. I suggest that in those cases you use an "M" (= mixed) instead of "R". Or you display the same frequency twice, once with "R" and once without.

2. Actually you catch the RV-information only for those frequencies that are within the AF-list for the tuned frequency. This is correct so far. But the AF-list on the TEF is then filled with ALL other AFs that are transmitted, also with all AFs that are not within the AF-list for the tuned frequency but also within all the other AF-lists for different frequencies. To make it right, you should do it with a further step:
a) First collect the complete AF-list of only (!) the tuned frequency -> display "R"- and "non-R"-frequencies
b) Then add all collected frequencies of the other AF-lists (all those that are NOT within the AF-list of the tuned frequency!). Display those AFs with an "O" (= other AF-list). This is important as you are not able to decide, if those frequencies are RV or not to the tuned frequency. For example: WDR 2 transmits the following AF-list for 100,4:
_[#17, 100,4 MHz] 101,0 MHz, 100,4 MHz (RV); 100,8 MHz, 100,4 MHz (RV); 98,6 MHz, 100,4 MHz; 100,4 MHz, 99,2 MHz (RV); 100,4 MHz, 95,7 MHz (RV); 88,4 MHz, 100,4 MHz; 91,8 MHz, 100,4 MHz; 100,4 MHz, 87,8 MHz (RV); 100,4 MHz, 99,8 MHz (RV); 100,4 MHz, 91,8 MHz (RV); 100,4 MHz, 93,5 MHz (RV)_
With this list it is possible to decide for "R", "non-R" and "M". The other AF-lists somewhere contain the further frequencies 92,3/93,2/93,3/93,8/94,1/94,2/96,0/96,4/97,1/99,1/99,4/101,8/102,1, those should be marked with "O".

(Of course "M" and "O" may differ if you prefer a different shortage...)

PE5PVB, 2023-10-13:
OK. Let me understand this. So you want to have:

Example. You receive (#19, 100,4 MHz] 101,0 MHz, 100,4 MHz (RV); 100,4 MHz, 100,8 MHz;, 100,8 MHz, 100,4 MHz (RV); 98,6 MHz, 100,4 MHz; 100,4 MHz, 99,2 MHz (RV); 100,4 MHz, 95,7 MHz (RV); 88,4 MHz, 100,4 MHz; 91,8 MHz, 100,4 MHz; 100,4 MHz, 87,8 MHz (RV); 100,4 MHz, 99,8 MHz (RV); 100,4 MHz, 91,8 MHz (RV); 100,4 MHz, 93,5 MHz (RV)

- All frequencies with RV are marked with R. 
- When a frequency also is shown as non-RV like 100.8 it is shown as M. 
- All others are shown without a remark
- All AF frequencies received on other tables are marked as O.

Correct?

lxsxl, 2023-10-13:
Yes, that's correct. Imho that's the best way to implement it on the small screen of the TEF. Where "All AF frequencies received on other tables" means all frequencies without those that are already caught from the list for the tuned frequency!

Btw and only to be sure we don't forget any side-effects:
1. This algorithm should not harm if method B is used an no RV is within the AF-list of the tuned frequency (so all AF of the tuned frequency will be without a remark, all further AF are marked with "O")
2. If AF-function is activated, what will you do with the AF-list? It would be correct to always rebuild the list because it may differ for each frequency.

PE5PVB, 2023-10-13:
1. Correct. At the time I detect the Method B I have to clear the full list and then flag all frequencies. Otherwise the normal AF function won't work correct.
2. How should AF act when in method B? Please advice.

lxsxl, 2023-10-13:
AF should act like it already does. As you respect the complete PI-Code it is quite fine. But if the AF function switches to a different frequency, you should also rebuild the AF table. But maybe it is already done like this...

PE5PVB, 2023-10-13:
No it's not. But sure a good idea that after AF retune in method-B the AF list should be cleared.

lxsxl, 2023-10-13:
I added a few lines to my upper post, maybe this is also interesting?

PE5PVB, 2023-10-13:
Please do not add something to earlier posts, but make a new post, because the whole thing is already a mindbreaker ;-)

PE5PVB, 2023-10-13:
But just to be clear. In AF method B all frequencies will have a marking.

R = Regional
M = Mixed
O = Other

There are no other possibilities. Right?

PE5PVB, 2023-10-13:
Another question for you for AF tuning. Is it correct that the 2nd nibble of the PI code can be different on AF-method B programs? If so, I have to update the AF routine, and can also add AF-REG tuning.

PE5PVB, 2023-10-13:
Tested it on WDR2. The M indicator works. I still have some questions about the Other flag. I think it's better not to show that, because all frequencies besides R and M will show a O, which is messy on the screen.

lxsxl, 2023-10-13:
To the AF indicators: There will still be at least a few frequencies without an indicator, namely those frequencies within the AF-list of the tuned frequency that transmit the same regional version. In the attached file you can see all the possible variants.

But you are right that most of all AFs will be with an "O". So it would be less messy if we turn over the logic : no indicator means "AFs from other lists" and let's say "S" for "same region" (formerly without an indicator). For WDR 2 with really large lists it fits better like this, but it may be that it is vice versa for different stations.

Another idea would be to skip all other lists but the one of the tuned frequency, but I think other users might miss them?

[D392 WDR 2 100,4 AF-lists.pdf](https://github.com/PE5PVB/TEF6686_ESP32/files/12899689/D392.WDR.2.100.4.AF-lists.pdf)

PE5PVB, 2023-10-13:
Can you give an example of which one should be marked as S in your PDF?

lxsxl, 2023-10-13:
To AF-tuning:
1. REG ON should switch only to the region-identical (= ascending order in the AF-list of the tuned frequency AND completely the same PI-code). If there is no such AF available, then switch to a receivable RV-AF (= descending order in the AF-list of the tuned frequency and PI-code with likely differing second nibble; second nibble may be from 3 to F (also important for AF-scan and displaying AFs in green font-colour!); 0 to 2 are for single frequencies, international or national programs and therefore not to be used to indicate regional versions). If there are neither region-identical AFs nor RV-AFs then you may also switch to AFs from other lists (as far as they are receivable).
2. REG OFF: Don't care for the second nibble of PI-code (as far as it is from 3 to F) and switch to one of the receivable AFs. If "REG OFF" is selected, than you can also skip all the calculation and indication for R/M/O/S (if you like to integrate this as a variant)

lxsxl, 2023-10-13:
A few further thoughts:
1. To make the AF view less messy: use two dots instead of the character: left dot means „same region as tuned frequency“, right dot means „RV“, both dots mean „mixed“, no dot means „other AF-lists“. The dots must be „column-aligned“ (hope you know what I mean?), so one can directly see, if an AF is the same or an RV.
2. AF-tuning in REG ON-mode can of course also be done as long as PI-code is the same (particularly when the central program is transmitted or if AF-order is done wrong by the broadcaster).
3. The „AF-R“ label better should be „AF-B“ because there are a lot of stations using AF-method B without RV.

lxsxl, 2023-10-14:
> Can you give an example of which one should be marked as S in your PDF?

"S" or "same-region-dot" are 88,4 and 98,6 (the ones in bold font in the 100,4-list).

PE5PVB, 2023-10-14:
88,4 MHz, 100,4 MHz

Ah ok, because 88.4 < 100.4 but there is no 100,4, 88,4MHz in the list?

lxsxl, 2023-10-14:
Yep, that‘s Right, cause 88,4 is the same version as 100,4. If there was also a descending 100,4/88,4 it would be an „M“ (which applies only to 91,8 in the 100,4-list).

PE5PVB, 2023-10-14:
Ah. Right. I just implemented that. I also changed the allignment of the S, M and R, and changed the color so it's easier to view.

PE5PVB, 2023-10-14:
The idea with the dots is good, but there is simply not enough room on the screen. Hopefully this will make is more clear to watch.

lxsxl, 2023-10-15:
First view on 100,4 shows the awaited behaviour. I will test other frequencies in comparison with RDS Spy to confirm it and let you know.

Btw: In the update-info you wrote "extra TMC-probe". What is meant with this? Actually for me it seems that with appearence of the first group 8A the TMC-indicator is shown, right? Or what else do you do on TMC?

PE5PVB, 2023-10-15:
Well, in the Netherlands when no traffic info is available the dedicated group is not transmitted, but the AID reference is. In that case TMC flag was not working.

lxsxl, 2023-10-15:
Ah, okay. I have never looked what happens on german stations if there are no TMC-messages.

lxsxl, 2023-10-15:
Now I checked a few further frequencies, and there are still not all indicators as aspected:
1. The "R"-AF were completely detected (and indicated).
2. The "S"-AF are only detected to a small extent. And the same way there are AFs that are expected as "M" but indicated as "R" (so also the "S"-component is missing?).
3. I've sent you some MPX-files in about one month ago, hope you still have it? You can test with the following frequencies:
D392 / WDR2 / 91,8 -> TEF shows "R" for 93,5 / 97,1 / 102,1 (all correct) as well as for 98,6 / 99,2 / 100,4 (should be "M"); no "S" at all (should be for 93,2/ 95,7 / 96,4 / 99,1 / 99,8 / 101,8, they all are with no indicator)
D392 / WDR2 / 93,5 -> TEF shows "R for 87,8 / 91,8 / 93,2 / 94,1 / 95,7 / 98,6 / 99,2 / 99,8 / 100,4 / "S" for 92,3 (correct but but should be also for 93,8 / 97,1 / 99,4 / 101,8 / 102,1); no "M" (which is correct)

I will go on with tests in the afternoon. And if you need them again I can send you the MPX-files

PE5PVB, 2023-10-15:
Found it. Silly mistake after copy-paste.

PE5PVB, 2023-10-15:
I've updated the repository, please check.

lxsxl, 2023-10-15:
Sorry, took me a while this time but now I was successful to compile on my own for the first time 😊
RV is has become better. Now all the AFs from the list for the tuned frequency has an indicator. But „M“ is not working, the expected frequencies have either „R“ or „S“.

PE5PVB, 2023-10-15:
Hmm. I think we have a bug in RDSSpy here. I made a script to output AF data from the TEF. The only time I found 91.8  100.4 was in the 100.4 table. See raw output:

#7	9350
	9350	9940
	9350	9380
	9350	9180
	
#15	9410
	9410	9600
	9410	9330
	9920	9410
	9410	9350
	9410	9320
	9410	8780
	9910	9410
	
#7	9420
	9420	10080
	9420	10100
	9920	9420
	
#5	9330
	9330	9920
	9410	9330
	
#7	9640
	10210	9640
	9320	9640
	9180	9640
	
#9	10100
	10080	10100
	10100	10040
	10100	9920
	9420	10100
	
#13	10180
	9350	10180
	10180	10210
	9710	10180
	9230	10180
	9380	10180
	9180	10180
	
#13	10080
	10080	10040
	10080	10100
	10080	9860
	10080	9920
	9420	10080
	10080	8840
	
#15	10210
	10180	10210
	10210	9640
	9350	10210
	10210	9320
	9230	10210
	9940	10210
	10210	9180
	
#17	10040
	10080	10040
	10100	10040
	9860	10040
	10040	9920
	10040	9350
	10040	9570
	8840	10040
	9180	10040
	
#7	9600
	9410	9600
	9600	9320
	9600	9910
	
#7	9710
	9710	10180
	9350	9710
	9230	9710
	
#9	8840
	10080	8840
	8840	10040
	8840	9860
	9920	8840
	
#25	9920
	10080	9920
	10040	9920
	10100	9920
	9330	9920
	9920	9860
	9920	9410
	9920	9350
	10210	9920
	9920	9570
	9920	9320
	9980	9920
	9920	8780
	
#5	9920
	9940	9920
	9180	9920
	
#5	9380
	9380	10180
	9350	9380
	
#7	9940
	9940	9920
	9350	9940
	9940	10210
	
#11	8780
	9920	8780
	9410	8780
	9350	8780
	10040	8780
	9980	8780
	
#11	9980
	10040	9980
	9980	9920
	9980	9350
	9570	9980
	9180	9980
	
#21	9320
	9320	9640
	9600	9320
	9920	9320
	9410	9320
	9350	9320
	10210	9320
	9320	8780
	9320	9910
	9180	9320
	8840	9320
	
#9	9230
	9230	10180
	9230	9350
	9230	10210
	9230	9710
	
#23	9180
	10040	9180
	9860	9180
	9920	9180
	9350	9180
	10210	9180
	9710	9180
	9180	9980
	9180	9570
	9180	9320
	9180	9640
	9180	9910
	
#11	9860
	10080	9860
	9860	10040
	9920	9860
	9860	9350
	9180	9860
	
#7	9910
	9910	9600
	9910	9410
	9320	9910
	
#9	9570
	10040	9570
	9920	9570
	9570	9980
	9180	9570
	
#25	9350
	10040	9350
	9350	10180
	9860	9350
	9920	9350
	9410	9350
	9350	10210
	9570	9350
	9350	9710
	9350	9320
	9230	9350
	9980	9350
	9350	8780
	
#7	9350
	9350	9940
	9350	9380
	9350	9180
	
#15	9410
	9410	9600
	9410	9330
	9920	9410
	9410	9350
	9410	9320
	9410	8780
	9910	9410
	
#7	9420
	9420	10080
	9420	10100
	9920	9420
	
#5	9330
	9330	9920
	9410	9330
	
#7	9640
	10210	9640
	9320	9640
	9180	9640
	
#9	10100
	10080	10100
	10100	10040
	10100	9920
	9920	9320
	9410	9320
	9350	9320
	10210	9320
	9320	8780
	9320	9910
	9180	9320
	8840	9320
	
#9	9230
	9230	10180
	9230	9350
	9230	10210
	9230	9710
	
#23	9180
	10040	9180
	9860	9180
	9920	9180
	9350	9180
	10210	9180
	9710	9180
	9180	9980
	9180	9570
	9180	9320
	9180	9640
	9180	9910
	
#11	9860
	10080	9860
	9860	10040
	9920	9860
	9860	9350
	9180	9860
	
#7	9910
	9910	9600
	9910	9410
	9320	9910
	
#9	9570
	10040	9570
	9920	9570
	9570	9980
	9180	9570
	
#25	9350
	10040	9350
	9350	10180
	9860	9350
	9920	9350
	9410	9350
	9350	10210
	9570	9350
	9350	9710
	9350	9320
	9230	9350
	9980	9350
	9350	8780
	
#7	9350
	9350	9940
	9350	9380
	9350	9180
	
#15	9410
	9410	9600
	9410	9330
	9920	9410
	9410	9350
	9410	9320
	9410	8780
	9910	9410
	
#7	9420
	9420	10080
	9420	10100
	9920	9420
	
#5	9330
	9330	9920
	9410	9330

lxsxl, 2023-10-16:
TEF is definitely the new reference ;-)
I think what happens in RDS Spy in method B: It is not only showing the AFs that are transmitted within one list but it adds further pairs from other AF-lists that contain this specific frequency. If there are differences in the assignement to R or S it will be shown twice.
So I would say keep it like it is, as it is most accurate now!
And at this moment I think you can close this issue :-)

andimik, 2023-10-16:
Regarding Rdsspy: Are fillers (RV) also allowed in method A?

lxsxl, 2023-10-17:
Fillers are not allowed in method B.

lxsxl, 2023-10-17:
Sorry, I was confused about your question. Fillers ARE allowed in method A but Not in method B. RV are possible in method B but not in method A.

PE5PVB, 2023-10-17:
AF Method B code cleanup finished. Please check if I didn't break anything ;-)

PE5PVB, 2023-10-18:
I'll close this issue.

closed 2023-10-18 by PE5PVB

lxsxl, 2023-10-18:
Sorry, I was not able to test until tonight. I used the penultimate version (today 17:30). As far as I  tested (AF-REG-Tuning is still open) I didn‘t find any error. So I would say, code-cleanup was successful. Great work 😊!

Closed #304 as completed.