PE5PVB/TEF6686_ESP32 #316: Different frequencies of the OIRT range are not saved

Yurij20, 2023-10-15:
Different frequencies of the OIRT range are not saved
https://youtu.be/ozmtukSKh3k?si=b9HkwH9AfootlQ9z

andimik, 2023-10-15:
Have you done a complete reset first?

PE5PVB, 2023-10-15:
Thanks. Fixed the problem.

Yurij20, 2023-10-16:
I checked.  On the OIRT band everything is saved normally, but on the LW, MW, SW bands there are still problems: it remembers the frequencies themselves but displays them incorrectly.  In the video below, I remember 2 frequencies of each range and switch them back and forth.  Here everything is clearly visible:
https://youtu.be/dIp1Nvrmwk4?si=yww2uckqx5iST9UP

PE5PVB, 2023-10-17:
Fixed that one.

Yurij20, 2023-10-18:
Thank you) everything is fixed.  The interference from the display on the receiver has also disappeared.  So please compile new firmware releases in the format that was compiled today)

closed 2023-10-18 by HyperDX